theori-io/nrsc5 #2: Failed to submit transfer 31! error

jeffgus, 2017-06-13:
I haven't had issues with other rtlsdr programs. rtl_fm, gqrx, and sdrtrunk all work. 
Does anyone have any ideas on how to get nrsc5 past this error?

Exact sample rate is: 1488375.071248 Hz
Failed to submit transfer 31!
rtlsdr error: -5

jeffgus, 2017-06-13:
I think I found the issue:
        -err = rtlsdr_read_async(dev, input_cb, &input, 0, RADIO_BUFFER);
        +err = rtlsdr_read_async(dev, input_cb, &input, 16, RADIO_BUFFER);

awesie, 2017-06-13:
Thanks for reporting the problem. I'm happy to commit that change, but passing 0 should've used a default value (in rtl-sdr) of 15. Can you check whether the issue exists with a value of 15?

jeffgus, 2017-06-13:
Oh! I'm fairly certain that I initially put 15, but then accidentally incremented the number in vim. I reduced it to 15 and re-built it and it works fine.

commit 7fc2ef20 (2017-06-13): Set rtl-sdr buffer count to default. Fixes #2.

Patch courtesy of @jeffgus.
closed 2017-06-13 by awesie