theori-io/nrsc5 #4: Compiling for ARM

BrodieBruce901, 2017-06-14:
I was very excited to hear about your work and decided to try compile the program. I got through most of the compile until I ran into this error gcc: error: unrecognized command line option '-msse2' and realized it's looking for an x86 based instruction set. I'm going to compile it on an old x86 laptop, but I was wondering if there is a possible work around or plans for an ARM compatible build?

awesie, 2017-06-14:
I committed a change to remove the -msse2 option from the Makefile, since it isn't necessary anyways. Does that fix compilation on ARM?

closed 2017-06-14 by awesie

BrodieBruce901, 2017-06-14:
I was just about to test it with the msse2 and --ffast-math flagw unset. I am about to try it. --ffast-math may also be an issue but I will let you know after the compile.

BrodieBruce901, 2017-06-14:
Not sure why I got a -ffast-math error earlier but it did compile without errors with the -msse2 flag unset from the prefix. Thanks for the project, the prompt response, and the help.

commit bd3b6fcc (2017-06-14): Remove sse2 compile option. Fixes #4.