theori-io/nrsc5 #5: Outputing the provided sample file or any capture to wav produces an unplayable file.

BrodieBruce901, 2017-06-14:
I unpacked the sample file and ran it through the software with the output to wav option. It produced a file 68 bytes in size that is unplayable. I also captured samples of a strong local signal. The local signal produced a 103MB sized file, but when I tried to output the sample to wav it produced a file 68 bytes in size that was also unplayable. This is happening on the recent ARM build that succeeded after removing the -msse2 flag.

BrodieBruce901, 2017-06-14:
Here is the output from the sample file on ARM
nrsc5 -o sample.wav -f wav -r sample 0
avg: 108.187500, slope: 0.031250, freqerr: 96.755287
avg: 109.062500, slope: 0.031250, freqerr: 93.589371
avg: 109.375000, slope: -0.093750, freqerr: 94.064873
avg: 109.375000, slope: 0.093750, freqerr: 93.403366
avg: 109.375000, slope: -0.031250, freqerr: 91.137932

awesie, 2017-06-14:
The issue was that char is unsigned on some platforms, but I relied on it being signed. Fixed by using signed char explicitly.

closed 2017-06-14 by awesie

BrodieBruce901, 2017-06-14:
That seemed to fix the issue but I'm not sure this ARM machine will be capable of decoding any live signals without multi thread support as nrsc5 is utilizing a core to 100% and it seems to be lagging decoding the blocks.

commit e9af39b1 (2017-06-14): Use signed char instead of char. Fixes #5.