theori-io/nrsc5 #6: Make fails while Building C object sync.c.o

BrodieBruce901, 2017-06-14:
While compiling on x86_64 with Ubuntu 14.04 with up to date packages the following errors occur with make
[ 88%] Building C object src/CMakeFiles/nrsc5.dir/sync.c.o
/user/nrsc5/src/sync.c: In function ‘dump_ref’:
/user/nrsc5/src/sync.c:25:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int i = 1; i < 32; i++)
     ^
/user/nrsc5/src/sync.c:25:5: note: use option -std=c99 or -std=gnu99 to compile your code
/user/nrsc5/src/sync.c: In function ‘calc_phase’:
/user/nrsc5/src/sync.c:36:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int r = 0; r < N; r++)
     ^
/user/nrsc5/src/sync.c:41:14: error: redefinition of ‘r’
     for (int r = 1; r < N; r++)
              ^
/user/nrsc5/src/sync.c:36:14: note: previous definition of ‘r’ was here
     for (int r = 0; r < N; r++)
              ^
/user/nrsc5/src/sync.c:41:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int r = 1; r < N; r++)
     ^
/user/nrsc5/src/sync.c: In function ‘adjust_ref’:
/user/nrsc5/src/sync.c:62:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int n = 0; n < N; n++)
     ^
/user/nrsc5/src/sync.c: In function ‘find_first_block’:
/user/nrsc5/src/sync.c:76:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int n = 0; n < N; n++)
     ^
/user/nrsc5/src/sync.c:83:14: error: redefinition of ‘n’
     for (int n = 0; n < N; n++)
              ^
/user/nrsc5/src/sync.c:76:14: note: previous definition of ‘n’ was here
     for (int n = 0; n < N; n++)
              ^
/user/nrsc5/src/sync.c:83:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int n = 0; n < N; n++)
     ^
/user/nrsc5/src/sync.c: In function ‘calc_smag’:
/user/nrsc5/src/sync.c:105:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int n = 0; n < N; n++)
     ^
/user/nrsc5/src/sync.c: In function ‘adjust_data’:
/user/nrsc5/src/sync.c:116:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int n = 0; n < N; n++)
     ^
/user/nrsc5/src/sync.c:122:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int k = 1; k < 19; k++)
         ^
/user/nrsc5/src/sync.c: In function ‘sync_process’:
/user/nrsc5/src/sync.c:175:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int n = 0; n < N; n++)
         ^
make[2]: *** [src/CMakeFiles/nrsc5.dir/sync.c.o] Error 1
make[1]: *** [src/CMakeFiles/nrsc5.dir/all] Error 2
make: *** [all] Error 2

awesie, 2017-06-14:
I add --std=gnu11 to the CFLAGS which should resolve this error. I also downgraded to CMake 2.8 so that it builds cleanly on my copy of Ubuntu 14.04.

closed 2017-06-14 by awesie

BrodieBruce901, 2017-06-14:
That fixed the problem. Thanks.

commit dddc3b46 (2017-06-14): Define CFLAGS in src, and test for autoconf.

Builds on Ubuntu 14.04. Fixes #6. This might fix #7.