theori-io/nrsc5 #7: Make fails while building liquid-dsp_external

RyokoAsakura, 2017-06-14:
Compile attempted on Kali4. All dependencies installed and current.
Also, thank you for your work.

```
[ 51%] Performing build step for 'liquid-dsp_external'
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/i686-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
/usr/lib/gcc/i686-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)
 ^
src/dotprod/src/dotprod_cccf.mmx.c:255:11: error: called from here
         s = _mm_addsub_ps( ci, cq );
           ^
<builtin>: recipe for target 'src/dotprod/src/dotprod_cccf.mmx.o' failed
make[3]: *** [src/dotprod/src/dotprod_cccf.mmx.o] Error 1
CMakeFiles/liquid-dsp_external.dir/build.make:111: recipe for target 'liquid-dsp-prefix/src/liquid-dsp_external-stamp/liquid-dsp_external-build' failed
make[2]: *** [liquid-dsp-prefix/src/liquid-dsp_external-stamp/liquid-dsp_external-build] Error 2
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/liquid-dsp_external.dir/all' failed
make[1]: *** [CMakeFiles/liquid-dsp_external.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

RyokoAsakura, 2017-06-14:
Fixed.
Changed CMakeLists.txt, line 5 to:
`set (CMAKE_C_FLAGS "-g -O2 -msse3 -ffast-math")`

Thank you again.

awesie, 2017-06-14:
Thanks for the report. I moved the CFLAGS definition to src/CMakeLists.txt so that it doesn't affect liquid-dsp. Hopefully that will fix the issue without adding the msse3 flag.

RyokoAsakura, 2017-06-14:
Deleted my build, attempted rebuild with fixes, build fails in the same place and with the same error as above. Just to let you know.

awesie, 2017-06-14:
Ok, thank you for testing it.

commit dddc3b46 (2017-06-14): Define CFLAGS in src, and test for autoconf.

Builds on Ubuntu 14.04. Fixes #6. This might fix #7.
commit eed9db92 (2017-06-14): Fix upstream issue with SSE3 support. Fixes #7.

Upstream should be using HAVE_SSE3 to test for SSE3 support instead of
HAVE_PMMINTRIN_H. Fork for now.
closed 2017-06-14 by awesie