theori-io/nrsc5 #8: Unknown HDC block type: 0

BrodieBruce901, 2017-06-14:
While trying to decode program 2 of a local station I receive the following error 

avg: 1092.500000, slope: -0.093750, freqerr: 66.531509
first block @ 7
synchronized!
avg: 1090.375000, slope: 0.031250, freqerr: 67.772034
BER: 0.013012, avg: 0.013012, min: 0.013012, max: 0.013012
pdu_seq: 0, seq: 61, nop: 33
ignoring partial pdu
Unknown HDC block type: 0
Segmentation fault (core dumped)

It is a low bandwidth talk sub channel if that is of any help.

BrodieBruce901, 2017-06-14:
I receive the same error with program 1 on the same frequency but not on program 0
avg: 199.312500, slope: -0.031250, freqerr: 41.423744
first block @ 27
synchronized!
avg: 197.375000, slope: 0.031250, freqerr: 38.220448
BER: 0.003883, avg: 0.003883, min: 0.003883, max: 0.003883
pdu_seq: 0, seq: 61, nop: 33
ignoring partial pdu
Unknown HDC block type: 0
Segmentation fault (core dumped)

awesie, 2017-06-14:
Interesting. Can you output it to a adts file (-o out.adts -f adts), zip it, and then attach to this issue? That should let me examine the packets.

BrodieBruce901, 2017-06-14:
[911-3.adts.gz](https://github.com/theori-io/nrsc5/files/1073838/911-3.adts.gz)
No error occurred while outputting the file.

awesie, 2017-06-14:
Thanks for the data; I hadn't encountered block type 0 before. It should be working now, though there may be some missing (stereo?) because there is some data in the packets that I am ignoring for now.

closed 2017-06-14 by awesie

BrodieBruce901, 2017-06-14:
Confirmed fixed. I believe these are mono streams so that shouldn't be an issue. Thanks for the hard work and quick turn around.

commit 983e211e (2017-06-14): Add support for HDC block type 0. Fixes #8.

Works with example HDC stream in #8 provided by @BrodieBruce901.