theori-io/nrsc5 #11: Experimental branch ARM

BrodieBruce901, 2017-06-21:
I am happy to report that the experimental branch compiled with no issues on ARM with the -DUSE_NEON=ON  option enabled. It does use significantly less CPU usage, however I am still not able to decode on the ARM machine. It automatically sets the gain but once that is finished all I get is repeating Timing offset: [offset value], slope: [slope value]. I am using the same sdr, station, and antenna placement my x86 computer is using that is able to decode with the experimental branch code compiled.

awesie, 2017-06-21:
I would try manually setting the ppm to a value that works on your x86 computer, and let me know if that fixes ARM. 

I've been testing the experimental branch on a Raspberry Pi 3 and my MacBook Pro, so I can say that it sometimes works on ARM. :)

BrodieBruce901, 2017-06-21:
I'll try that and report back.

BrodieBruce901, 2017-06-22:
So I have tried a few things and have some observations. You had asked for me to copy the ppm settings from my x86 machine but I have never used the ppm setting. It has been fine give or take for the things I could decode with the Master branch. With the Experimental branch I don't have to tweak anything on x86. Since I didn't have any ppm setting to go by I ran rtl_test -p for about 10 minutes. I have some wild variations. Here is a snippet:  

real sample rate: 2049264 current PPM: 618 cumulative PPM: 36
real sample rate: 2048058 current PPM: 29 cumulative PPM: 35
real sample rate: 2048064 current PPM: 31 cumulative PPM: 35
real sample rate: 2047375 current PPM: -305 cumulative PPM: 27
real sample rate: 2047743 current PPM: -125 cumulative PPM: 24
real sample rate: 2049045 current PPM: 510 cumulative PPM: 35
real sample rate: 2048090 current PPM: 44 cumulative PPM: 35
real sample rate: 2047048 current PPM: -465 cumulative PPM: 24
real sample rate: 2048628 current PPM: 307 cumulative PPM: 30
real sample rate: 2046587 current PPM: -690 cumulative PPM: 15
real sample rate: 2049936 current PPM: 946 cumulative PPM: 34
real sample rate: 2048100 current PPM: 49 cumulative PPM: 35
real sample rate: 2048058 current PPM: 29 cumulative PPM: 34
real sample rate: 2048060 current PPM: 30 cumulative PPM: 34
real sample rate: 2048059 current PPM: 29 cumulative PPM: 34
real sample rate: 2046137 current PPM: -910 cumulative PPM: 17
real sample rate: 2048966 current PPM: 472 cumulative PPM: 25
real sample rate: 2049080 current PPM: 527 cumulative PPM: 34

I went ahead and tried it using a setting of 34. I tuned to what is the strongest and easiest signal to decode on the x86 machine. Here is the thing though, when I had the Master branch compiled for the ARM machine I was able to detect blocks and sync. I would just immediately lose sync due to lack of resources. Now I don't get any block detection. I get this: 

Timing offset: 465.875000, slope: -2.375000 (adjust)

It also takes a long time to get that output or any subsequent messages like it and that's all it will do. It isn't a resource issue from what I can tell. The program is using a single core and it is not fully utilized. It runs somewhere around 40 to 50%, a drastic improvement from the previous build. As stated in another bug report, I am not able to decode the sample in the support folder now either. I created one from the x86 machine and I am about to test it on the ARM in case there was just something wrong with the support sample.

awesie, 2017-06-22:
Thanks for testing things.

I committed a couple of changes. Some of them fix potential issues that may have caused this behavior. In case it doesn't, I also added a USE_THREADS compile flag which you can try turning off just to see if it fixes the issue.

BrodieBruce901, 2017-06-22:
You're welcome. You're doing some amazing work with unreal turnaround. I started a Facebook group so those testing could share what they find, get or offer help. It can be found here https://www.facebook.com/groups/NRSC5SDR I plan on working with a Raspberry Pi 3b this weekend, but I have been concentrating on getting this Rockchip machine going so I can easily do field tests when I am on the road for work. I'm also planning on chasing some tropospheric ducting events based on forecasts this summer and having it working on the laptop is a big component to that. I did manage a 60+ mile sync on an indoor antenna with the x86 machine this morning running the new experimental code. I'm going to recompile and I'll let you know what happens.

BrodieBruce901, 2017-06-22:
100% confirmed working on an Rockchip 1.8GHz Quad-core RK3288C. You sir are genius.

closed 2017-06-22 by BrodieBruce901