theori-io/nrsc5 #13: Compiling on Cygwin

K2DLS, 2017-06-21:
Has anyone succeeded in compiling on Cygwin 64 bit?  I can compile with a number of warnings but the linker fails on some unresolved symbols, the first of which is R_X86_64_PC32.

BrodieBruce901, 2017-06-23:
Have you tried updating binutils and recompiling?

K2DLS, 2017-06-23:
I am running the latest available binutils package for Cygwin, but reinstalled just for good measure with same results.  I suspect this has something to do with Cygwin 32 vs 64 bits based upon the symbol mentioned above.

BrodieBruce901, 2017-06-23:
I've been doing some research to see if it is something that can be resolved on your end. Can you post  the complete error you get with the R_X86_64_PC32?

K2DLS, 2017-06-23:
[100%] Linking C executable nrsc5.exe
CMakeFiles/nrsc5.dir/input.c.o:input.c:(.text+0x2a6): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/input.c.o:input.c:(.text+0x2a6): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `CMPLXF'
CMakeFiles/nrsc5.dir/input.c.o:input.c:(.text+0x2f3): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/input.c.o:input.c:(.text+0x2f3): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `CMPLXF'
CMakeFiles/nrsc5.dir/sync.c.o:sync.c:(.text+0x4ac): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/sync.c.o:sync.c:(.text+0x4ac): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `CMPLXF'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `_libusb_callback':
/home/dan/git/rtl-sdr/src/librtlsdr.c:1703: undefined reference to `libusb_submit_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1703:(.text+0x34f): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_submit_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_read_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:413: undefined reference to `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:413:(.text+0x485): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_write_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:426: undefined reference to `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:426:(.text+0x4c8): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:426: undefined reference to `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:426:(.text+0x521): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:426: undefined reference to `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:426:(.text+0x57e): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_read_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:413: undefined reference to `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:413:(.text+0x5b4): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o):/home/dan/git/rtl-sdr/src/librtlsdr.c:426: more undefined references to `libusb_control_transfer' follow
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_write_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:426:(.text+0x608): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_read_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:413:(.text+0x658): additional relocation overflows omitted from the output
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_get_usb_strings':
/home/dan/git/rtl-sdr/src/librtlsdr.c:794: undefined reference to `libusb_get_device'
/home/dan/git/rtl-sdr/src/librtlsdr.c:796: undefined reference to `libusb_get_device_descriptor'
/home/dan/git/rtl-sdr/src/librtlsdr.c:802: undefined reference to `libusb_get_string_descriptor_ascii'
/home/dan/git/rtl-sdr/src/librtlsdr.c:809: undefined reference to `libusb_get_string_descriptor_ascii'
/home/dan/git/rtl-sdr/src/librtlsdr.c:816: undefined reference to `libusb_get_string_descriptor_ascii'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_write_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:426: undefined reference to `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_read_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:413: undefined reference to `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_write_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:426: undefined reference to `libusb_control_transfer'
/home/dan/git/rtl-sdr/src/librtlsdr.c:426: undefined reference to `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_read_array':
/home/dan/git/rtl-sdr/src/librtlsdr.c:413: undefined reference to `libusb_control_transfer'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_get_device_count':
/home/dan/git/rtl-sdr/src/librtlsdr.c:1309: undefined reference to `libusb_init'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1313: undefined reference to `libusb_get_device_list'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1316: undefined reference to `libusb_get_device_descriptor'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1322: undefined reference to `libusb_free_device_list'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1324: undefined reference to `libusb_exit'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `rtlsdr_get_device_name':
/home/dan/git/rtl-sdr/src/librtlsdr.c:1339: undefined reference to `libusb_init'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1343: undefined reference to `libusb_get_device_list'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1346: undefined reference to `libusb_get_device_descriptor'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1358: undefined reference to `libusb_free_device_list'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1360: undefined reference to `libusb_exit'
/home/dan/git/rtl-sdr/src/librtlsdr.c:1358: undefined reference to `libusb_free_device_list'
[... cut ...]
/home/dan/git/rtl-sdr/src/librtlsdr.c:1875: undefined reference to `libusb_handle_events_timeout'
/usr/local/lib/librtlsdr.a(librtlsdr.o): In function `_rtlsdr_alloc_async_buffers':
/home/dan/git/rtl-sdr/src/librtlsdr.c:1740: undefined reference to `libusb_alloc_transfer'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/nrsc5.dir/build.make:334: src/nrsc5.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:161: src/CMakeFiles/nrsc5.dir/all] Error 2
make: *** [Makefile:128: all] Error 2

BrodieBruce901, 2017-06-23:
Looks like you don't have the the dependencies met. You need these llibao-dev, libfftw3-dev, and rtl-sdr
requirements met before it can be compiled.

BrodieBruce901, 2017-06-23:
It looks like you don't have the dependencies met. Here is a similar issue also involving libfftw3-dev
[https://stackoverflow.com/questions/19501780/cygwin-64-library-undefined-reference-relocation-truncated-to-fit-r-x86-64-pc](url)
Make sure you have all the dependencies installed and then try to recompile.

K2DLS, 2017-06-23:
I am aware of the dependencies (I have built under x86_64 Debian and Fedora).  I installed libao-dev and libfftw3-dev from the packages and built rtl-sdr from git source.  The following libraries are found in the filesystem.  Cygwin lacks an ldconfig, so I don't see any obvious way to make sure that the linker can find them.

dan@shack /lib
$ ls libao*
libao.dll.a

$ ls libfft*
libfftw3.dll.a          libfftw3f.dll.a          libfftw3l.dll.a
libfftw3_omp.dll.a      libfftw3f_omp.dll.a      libfftw3l_omp.dll.a
libfftw3_threads.dll.a  libfftw3f_threads.dll.a  libfftw3l_threads.dll.a

dan@shack /usr/local/lib
$ ls librtl*
librtlsdr.a  librtlsdr.la

BrodieBruce901, 2017-06-23:
Yeah that would make a difference. I have some experience with cygwin but have never needed to compile anything for it.

pclov3r, 2017-08-01:
It does build with MinGW with some modifications to the CMakeLists to mostly set the GCC cross-compiler. libusb,libfftw,rtlsdr are all precompiled but you will need to build libao yourself. 

However, With the experimental branch you will need to implement strdup since it is not implemented in MinGW. I used the follow bit of code.

```
#ifndef strdup
/**
* return a malloc'd copy of at most the specified
* number of bytes of a string
* @author glibc-Team
*/
char *strndup (const char *s, size_t n)
{
  size_t len = strnlen (s, n);
  char *new = (char *) malloc (len + 1);

  if (new == NULL)
    return NULL;

  new[len] = '\0';
  return (char *) memcpy (new, s, len);
}
#endif 
```

The CMPLXF macro is not defined as well and this needs to be defined. 

`#define CMPLXF(x, y) __builtin_complex ((float) (x), (float) (y))`

Hopefully this helps.

awesie, 2017-08-02:
I'll add some wrappers for strndup and CMPLXF to the makefiles.

pclov3r, 2017-08-02:
@awesie While your at it with makefiles LDFLAGS=-no-undefined needs to be passed for FAAD to build and link correctly.

--host i686-w64-mingw32 or --host x86_64-w64-mingw32 needs to be added to the FAAD configure command as well.

For ID3V2LIB CMAKE_ARGS -D CMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE} I added as well.

This is the entire modified CMakeLists file that works for me on a 64-bit build. 

```
cmake_minimum_required (VERSION 2.8)
include (CheckLibraryExists)
include (ExternalProject)
project (nrsc5 C)

option (USE_COLOR "Colorize log output")
option (USE_NEON "Use NEON instructions")
option (USE_SSE "Use SSE3 instructions")
option (USE_THREADS "Enable multithreading" ON)
option (USE_FAST_MATH "Use unsafe math optimizations")
option (USE_FAAD2 "AAC decoding with FAAD2" ON)
option (USE_ID3V2LIB "ID3V2 decoding from PSD" ON)

find_program (AUTOCONF autoconf)
if (NOT AUTOCONF)
    message (FATAL_ERROR "Missing autoconf. Install autoconf package and try again.")
endif ()

find_program (AUTOMAKE automake)
if (NOT AUTOMAKE)
    message (FATAL_ERROR "Missing automake. Install autoconf package and try again.")
endif ()

find_program (LIBTOOLIZE NAMES libtoolize glibtoolize)
if (NOT LIBTOOLIZE)
    message (FATAL_ERROR "Missing libtoolize. Install libtool package and try again.")
endif ()

find_library (FFTW3F_LIBRARY fftw3f)
find_library (RTL_SDR_LIBRARY rtlsdr)

if (CMAKE_SYSTEM_PROCESSOR MATCHES "arm.*")
    if (USE_NEON)
        set (CMAKE_C_FLAGS "-mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4")
        add_definitions (-DHAVE_NEON)
    endif()
endif()

if (CMAKE_SYSTEM_PROCESSOR MATCHES "(i[456]|x)86.*")
    if (USE_SSE)
        set (CMAKE_C_FLAGS "-msse2 -msse3 -mssse3")
        add_definitions (-DHAVE_SSE2 -DHAVE_SSE3)
    endif()
endif()

if (USE_FAAD2)
    # libao only used if we have FAAD2
    find_library (AO_LIBRARY ao)

    set (FAAD2_PREFIX "${CMAKE_BINARY_DIR}/faad2-prefix")
    ExternalProject_Add (
        faad2_external
        GIT_REPOSITORY "https://github.com/dsvensson/faad2.git"
        GIT_TAG b7aa099fd3220b71180ed2b0bc19dc6209a1b418
        PREFIX ${FAAD2_PREFIX}

        UPDATE_COMMAND ""
        PATCH_COMMAND patch -p1 -i "${CMAKE_SOURCE_DIR}/support/faad2-hdc-support.patch"
        COMMAND sh ./bootstrap

        CONFIGURE_COMMAND ${FAAD2_PREFIX}/src/faad2_external/configure --host x86_64-w64-mingw32 --prefix=${FAAD2_PREFIX} --with-hdc "CFLAGS=-O3 ${CMAKE_C_FLAGS}"

        BUILD_COMMAND make LDFLAGS=-no-undefined
    )

    add_library (faad2 STATIC IMPORTED)
    set_property (TARGET faad2 PROPERTY IMPORTED_LOCATION "${FAAD2_PREFIX}/lib/libfaad.a")
    add_dependencies (faad2 faad2_external)
    include_directories ("${FAAD2_PREFIX}/include")

    set (FAAD2_LIBRARY faad2)
    add_definitions (-DHAVE_FAAD2)
endif()

if (USE_ID3V2LIB)
    set (ID3V2LIB_PREFIX "${CMAKE_BINARY_DIR}/id3v2lib-prefix")
    ExternalProject_Add (
        id3v2lib_external
        GIT_REPOSITORY "https://github.com/larsbs/id3v2lib.git"
        GIT_TAG 9bca1c33b8cd8037ee59de927bda057c9ac7043d
	CMAKE_ARGS -D CMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}
        PREFIX ${ID3V2LIB_PREFIX}

        INSTALL_COMMAND ""
    )

    add_library (id3v2lib STATIC IMPORTED)
    set_property (TARGET id3v2lib PROPERTY IMPORTED_LOCATION "${ID3V2LIB_PREFIX}/src/id3v2lib_external-build/src/libid3v2.a")
    add_dependencies (id3v2lib id3v2lib_external)
    include_directories ("${ID3V2LIB_PREFIX}/src/id3v2lib_external/include")

    set (ID3V2LIB_LIBRARY id3v2lib)
    add_definitions (-DHAVE_ID3V2LIB)
endif()

add_subdirectory (src)

```
Unsure how to incorporate those changes.

awesie, 2017-08-02:
Compilation on Windows fixed with 1d19244. Use MSYS2 and MinGW64.

closed 2017-08-02 by awesie

K2DLS, 2017-08-04:
I've been following the discussion.  I am not clear as to whether my reported issue is also addressed by the ability to compile under MinGW?  Should this now compile under Cygwin (the issue that I originally raised) or must I move to an alternate environment (MinGW with MSYS) to be able to run under Windows x64?

awesie, 2017-08-04:
Sorry, I had equated "compiling on cygwin" to "compiling on Windows". If it still does not compile under cygwin, feel free to reopen the issue.