theori-io/nrsc5 #25: Parse data from AAS PDUs

awesie, 2017-07-11:
I wrote some code, that I'll commit in a bit, to parse the fixed data portion of the L2 PDUs. The result is a bunch of packets that I'm not quite sure what to do with.

```
unknown PSD port 411, seq 114b, length 269
08 01 57 99 0C 00 00 00 53 7B 26 93 EC EC 34 F2 FF 00 1B 22 FA 13 FA AB 94 37 F3 35 CD EE 93 A3 F2 87 CF EF 9D FB 24 13 BF 37 60 1C 09 7B 80 3F DA 39 7B CB BF DA BD CF DA 73 CD 5F F8 58 7E 6E C3 AF A2 B7 55 C5 E4 AC 4F 07 C1 70 D9 4B 7E 91 9A 7F E5 2F 9F 4E CF F4 E7 1D D2 AF 8F F8 1E BE AA B7 FA 90 9E F8 D3 C4 FE D2 C7 5A 23 2A 10 04 01 00 40 10 04 01 00 40 10 04 01 00 40 10 1A 6D
unknown PSD port 404, seq 2957, length 652
C9 3F 95 48 A7 2E 85 34 5B 24 63 09 E4 9E 75 AF 2A 24 10 A9 EC AC 59 DE 68 ED 37 5A 09 64 3E D6 60 B9 92 72 AD 0A DA 95 99 63 6E 55 27 A3 9D 80 5B 82 2C 54 2C 8C C7 4A 4E F2 66 B2 2A 15 25 F0 57 88 1A 54 37 B6 44 10 1B 6F 6C 56 4D 54 2C 8A 44 29 96 34 93 64 74 90 9C EB E5 68 54 8E A2 48 EE 8A 45 B3 5B 36 22 48 3E 39 19 4F 20 55 91 E4 15 CB CA 6C 71 3C 90 D1 5D 6F D5 DE EF 23 7C B4
unknown PSD port 402, seq 298d, length 818
40 F2 EF 46 99 3A 65 FC 98 E6 21 FE DE 13 32 7F 97 77 12 BC 4B F0 CD 3A 5F 7C 08 CD 2E A7 0D 87 69 83 B9 70 17 F7 C9 57 CD A4 F1 6F 82 77 CA 07 60 44 09 94 42 25 90 53 2A FD 32 D4 4A FD 7B A8 04 72 CA B3 CE 14 E2 1D 54 2B 72 CA 57 DF 65 C4 D7 2E 09 A1 DF FD 15 3A 00 4E B5 5E 8D CF 83 FD 4E 07 D4 3D 13 D0 E3 19 E7 88 5A 76 08 43 34 B3 3F 18 51 17 4A C3 F0 50 46 20 B4 FA 23 E8 AE 93
unknown PSD port 401, seq 2364, length 933
52 02 00 A0 40 00 0D 14 EF 00 00 00 04 80 B9 D0 A6 95 47 FA 40 00 0D 14 EF 00 00 00 04 80 B9 CF 4E 95 48 26 40 00 0D B1 0D 11 70 00 04 80 B9 D5 86 95 3F 50 40 E5 0D B0 00 00 00 00 05 40 B9 D7 0A 95 38 72 40 E5 0D B0 FC 11 B0 00 04 80 B9 D0 22 95 35 D4 40 E5 0D 14 F9 11 20 00 00 E8 B9 D7 12 95 37 86 40 00 0D 10 00 51 70 00 04 80 B9 D2 BA 95 32 A5 44 DB 0C E4 F9 51 70 00 01 A8 B9 D0
unknown PSD port 20, seq 9a2f, length 407
40 01 00 62 69 05 00 4D 50 53 66 0C 00 00 07 00 00 00 00 5A 6C C6 4D 67 0D 01 07 04 09 01 03 00 00 36 75 4B BE 60 09 53 65 6C 66 28 00 00 00 67 0D 02 11 04 09 01 03 00 00 36 25 C7 D9 60 09 53 65 6C 66 32 00 00 00 40 02 00 02 69 06 00 53 50 53 31 66 0C 00 01 00 00 00 00 00 5A 6C C6 4D 40 03 00 03 69 06 00 53 50 53 32 66 0C 00 02 00 00 00 00 00 5A 6C C6 4D 41 06 00 3B 69 12 00 4E 61
unknown PSD port 411, seq 114c, length 269
08 01 57 99 0D 00 00 00 13 70 97 83 9C 13 4A 2B E9 D0 E1 F8 8A 9F EA 1D 1D DC B7 37 D8 5E FD 2D A7 BD 86 F4 BB 7C E4 CA F0 8B BD C5 8E 80 71 8E D2 39 6F 9F E4 D4 E1 E0 43 F0 87 C8 7D 4A E2 E7 A7 7A 2B 64 53 97 62 EB 65 15 AF 42 C4 E5 B6 4D 47 DE 97 54 7C 66 E1 4C 20 15 CE C8 FF 00 27 DF D9 FC 11 EE C5 39 FA C5 BB 7B 28 0E B6 81 F7 92 FB 16 77 2D FD 1C 5D CB 5B 1E 95 E1 CC CD 56 6D
```

awesie, 2017-07-11:
I was able to kind of decode port 20, which apparently describes the other ports:
```
0140 6200
     69 '\x00MPS'
     66 000007000000005a6cc64d
     67 (0407) 010704090103000036754bbe
     60 53656c6628000000
     67 (0411) 02110409010300003625c7d9
     60 53656c6632000000
0240 0200
     69 '\x00SPS1'
     66 000100000000005a6cc64d
0340 0300
     69 '\x00SPS2'
     66 000200000000005a6cc64d
0641 3B00
     69 '\x00NavteqPacketData'
     67 (0401) 00010441000100003eac422d
     60 4e56543101000000
0841 9200
     69 '\x00NavteqAdmin'
     67 (0403) 00030407010100003eac422d
     60 4e56543103000000
0A41 E100
     69 '\x00AlbumArt2'
     67 (0408) 000804090103000036754bbe
     60 53656c6629000000
     67 (0409) 010904090103000036754bbe
     60 53656c662a000000
     67 (0412) 02120409010300003625c7d9
     60 53656c6633000000
     67 (0413) 03130409010300003625c7d9
     60 53656c6634000000
1141 E000
     69 '\x00HERE TPEG'
     67 (0402) 0002040301000000fc3df082
     60 4845524514000000
1241 C000
     69 '\x00HERE Images'
     67 (0404) 0004040301000000fc3df0b7
     60 4845524515000000
     CRC? E1
```

awesie, 2017-07-15:
Basic code to parse out files are now in experimental (12d9526f). Use the "--dump-aas-files" option. The "Navteq" and "HERE" data is not parsed currently.

iHeartMedia stations appear to send out files for both traffic and weather. TMT*.png are the map tiles (images). DWR0*.png is the weather overlay image. TMI*.txt and DWRI*.txt provide metadata for the maps.

vielmetti, 2022-11-17:
Michigan Radio (WUOM-HD1 Ann Arbor, 91.7 FM) has some kind of "Navteq" and "HERE" data. It's an NPR station.

A partial log:

```
22:56:48 SIG Service: type=data number=10 name=NavteqPacketData1
22:56:48   Data component: id=0 port=0401 service_data_type=65 type=1 mime=2D42AC3E
22:56:48 SIG Service: type=data number=12 name=NavteqAdmin
22:56:48   Data component: id=0 port=0403 service_data_type=263 type=1 mime=2D42AC3E
22:56:48 SIG Service: type=data number=22 name=HERE TPEG
22:56:48   Data component: id=0 port=0402 service_data_type=259 type=0 mime=82F03DFC
22:56:48 SIG Service: type=data number=23 name=HERE Images
22:56:48   Data component: id=0 port=0404 service_data_type=259 type=0 mime=B7F03DFC
```

It also spits out LOT files:

```
23:35:23 LOT file: port=0411 lot=17 name=Car Radio Logo_gray 2.jpg size=5765 mime=1E653E9C
```

[image]

argilo, 2025-05-07:
I'll close this off now, since all three AAS data types (stream, packet, LOT) are now processed and exposed via the API.

Higher-level decoding of specific data services can be tracked in individual issues like #401.

closed 2025-05-07 by argilo