theori-io/nrsc5 #33: AM NRSC-5

pclov3r, 2017-07-17:
Would it be possible to add support for the AM Broadcast band? From what I understand the sidebands are much more narrow and use much higher order modulation.

argilo, 2017-07-17:
To start, we'd need a good recording of an AM broadcast. Do you have any local stations?

pclov3r, 2017-07-17:
Unfortunately, I don't.

I can get a station via skip that is around 400 miles away but the sidebands do have fading.  Due to the higher order modulation this may not work well.

argilo, 2017-07-17:
Well, a recording with some fading is better than no recording at all. If you can capture a couple minutes of RF and post it somewhere, that would definitely help.

pclov3r, 2017-07-17:
Actually, The Signal ID wiki has a I/Q recording http://www.sigidwiki.com/wiki/File:AM_HD_Radio_iBiquity.zip

It does appear to be a weaker SNR but stable and may work.

argilo, 2017-07-17:
That could be useful as a starting point. Having a longer recording would still be helpful, as that one is only 14 seconds.

argilo, 2017-07-18:
In the sigidwiki sample, the digital sidebands are only 7-8 dB above the noise floor, which I expect is much too low for the signal to be decoded. We'll need a much better recording to work with.

pclov3r, 2017-07-18:
That's what I figured. Hopefully I can find a better sample or somebody can create a better sample.

billybobgot, 2017-07-21:
Here is a 2 minute recording of my local station in DFW 1.080 MHz
https://drive.google.com/open?id=0B3IrMfVr0MxqazNyelZRLVY5Q1E
Recorded at 16 bit using SDR Console v2.3 build 2760
Was using my Sdr play RSP1
Message me if you need anything

argilo, 2017-07-21:
Thanks @billybobgot! That looks like a great recording. When I get some time, I'll take a shot at decoding it.

pclov3r, 2017-08-02:
@argilo Curious if you're had any time or got anywhere with this?

Thanks!

argilo, 2017-08-02:
As it happens, I'm playing with this right now. But layer 1 AM is quite different from layer 1 FM, so I expect it will be a while until I get it sorted out.

pclov3r, 2017-09-07:
Have you managed to figure out anything for this over the past month or so? 

Thanks!

argilo, 2017-09-07:
Unfortunately I haven't made much progress yet. Most of my free time in the past month was spent building an NRSC-5 transmitter: https://github.com/argilo/gr-nrsc5

I'll probably get back to AM eventually, but I can't promise when.

pclov3r, 2019-02-12:
@argilo Did you ever make any progress with this? 

I'd be willing to help test tho it would require a MW/HF up-converter for use with a RTL-SDR.

argilo, 2019-02-12:
Unfortunately I haven't made any progress yet. I'll let you know if I do.

pclov3r, 2020-06-27:
Is there still any plans to have AMBC NRSC-5 support? 

Now, This would require a modded RTl-SDR or a pre-modded SDR that can do direct sampling or a upconverter that would require a frequency offset. Also, Support for RTL-SDR's that can use a Bias-T would be good idea people that use loop antennas. 

Seems like it would require some extra work would be required to implement this well enough for people to use.

argilo, 2020-06-27:
I would still like to add AM support.

I made an attempt to add AM support to gr-nrsc5 (https://github.com/argilo/gr-nrsc5/compare/add-am) and was able to get the PIDS (station metadata) channel working, but I'm still stuck trying to get the P1 & P3 (audio) channels going. A couple things in the standard are unclear, so it might help to go in the other direction and decode a real signal to see exactly how the data is encoded.

argilo, 2020-07-20:
I finally got transmit working: https://github.com/argilo/gr-nrsc5/commit/48c2e59e843c390af352dbe9916a050e8528d0e8

I've started working on receive as well, but there's still a fair bit of work left because AM NRSC5 differs from FM in a lot of ways.

argilo, 2020-07-20:
The code and receiver design are quite messy at the moment, but I've at least managed to get far enough to decode PIDS:

https://github.com/theori-io/nrsc5/compare/master...argilo:add-am

```
23:53:01 Block @ 30
23:53:01 Sync!
23:53:01 Country: US, FCC facility ID: 59820
23:53:01 Station name: KRLD
23:53:01 Country: US, FCC facility ID: 59820
23:53:01 Station name: KRLD
23:53:01 Audio program 0: public, type: None, sound experience 0
23:53:01 Country: US, FCC facility ID: 59820
23:53:01 Station name: KRLD
23:53:01 Station location: 39.956055, -91.363403, 192m
23:53:01 Audio program 0: public, type: None, sound experience 0
23:53:01 Country: US, FCC facility ID: 59820
23:53:01 Station name: KRLD
23:53:01 Slogan: NewsRadio1080 KRLD
23:53:01 Station location: 39.956055, -91.363403, 192m
23:53:01 Audio program 0: public, type: None, sound experience 0
23:53:01 Country: US, FCC facility ID: 59820
23:53:01 Station name: KRLD
23:53:01 Slogan: NewsRadio1080 KRLD
23:53:01 Message: NewsRadio1080 KRLD
23:53:01 Station location: 39.956055, -91.363403, 192m
23:53:01 Audio program 0: public, type: None, sound experience 0
```

pclov3r, 2020-08-03:
Some nice progress. Looks like you got Audio working now yay!

Any ideas for features and things supported for AMBC? I figure to give best coverage support of a HF upconverter and direct sampling needs to be supported. Also an option to enable the Bias-T for people using loop antennas.

argilo, 2020-08-03:
Yes indeed, today I finally got to the point where audio decoding works. That said, there's a lot of things that still need to be improved. Here's my to-do list:

- [ ] Add CFO detection. Without it, the receiver won't lock unless the SDR is tuned within 90 Hz of the correct frequency.
- [ ] Decode the audio packets in the "Enhanced" audio stream. This will improve audio quality (20 kbps → 36 kbps).
- [ ] Implement k=9 convolutional decoders for SSE & NEON, or modify the code to always use the generic convolutional decoder when k=9. Currently the code does not compile if the `-DUSE_SSE=ON` or `-DUSE_NEON=ON` flags are used.
- [ ] Improve the multi-stage decimator. The AM receiver currently uses a cascade of five fixed-point decimators, and each one has a gain of two. The input needs to be attenuated to compensate for the gain of the decimators, which reduces the dynamic range. Either the decimators could be changed to unity gain, or they could be replaced with a floating-point implementation.
- [ ] Improve synchronization, in particular by adding per-subchannel tracking loops to average out noise in the training/pilot symbols.
- [ ] Replace magic numbers with named constants.
- [ ] Implement automatic gain control.

argilo, 2020-08-03:
If anyone has a local station, I would appreciate additional I/Q recordings. All I have to work with at the moment is @billybobgot's recording, and it seems to have some AGC artifacts (visible as vertical bands) which makes decoding a challenge:

[image]

argilo, 2020-08-03:
I'd especially love a recording of WWFD (Frederick, Maryland), which broadcasts in the all-digital format: https://en.wikipedia.org/wiki/WWFD.

pclov3r, 2020-08-03:
Does the recording need to be 8-bit or is 16-bit okay?

argilo, 2020-08-03:
16-bit would be best. I can easily convert to whatever format I need with GNU Radio.

pclov3r, 2020-08-03:
I am able to get a few HD AMBC stations during the night but i don't think this would be of use to you since they are received via skip and not local groundwave propagation and have lots of selective frequency fading.

argilo, 2020-08-03:
AM NRSC-5 should be somewhat tolerant of frequency-selective fading since it uses OFDM. Also, there are gaps at ±10 kHz where no OFDM data carriers are used, thereby avoiding interference from adjacent AM carriers. But the required signal-to-noise ratio is relatively high, so I expect reception via skip to be difficult. It might be possible when the primary sidebands are >20 dB above the noise.

pclov3r, 2020-08-03:
Yeah, I can look tonight to see how the signals actually are.

Would you mind sharing how to convert a 16-bit I/Q file from another program for use with the NRSC5 application using GNUradio for both AM and FM? I looked at it doing it in the past and wasn't sure exactly how to do it and scale things?. Might be useful for others as well. 

My DSP knowledge is severely lacking here :)

argilo, 2020-08-04:
I used the following GNU Radio 3.8 flow graph to convert @billybobgot's recording to the right format:

[image]

This lowers the frequency by 30 kHz so that the target signal is at 0 Hz, resamples it from 500,000 to 1,488,375 samples per second and writes out 8-bit complex unsigned samples. 

Here's the .grc file:
```
options:
  parameters:
    author: ''
    category: '[GRC Hier Blocks]'
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: ''
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: no_gui
    hier_block_src_path: '.:'
    id: am_hd_convert
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: run
    sizing_mode: fixed
    thread_safe_setters: ''
    title: Not titled yet
    window_size: ''
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [8, 12.0]
    rotation: 0
    state: enabled

blocks:
- name: samp_rate
  id: variable
  parameters:
    comment: ''
    value: '500000'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [200, 12.0]
    rotation: 0
    state: enabled
- name: blocks_add_const_vxx_0_0
  id: blocks_add_const_vxx
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    const: '127.5'
    maxoutbuf: '0'
    minoutbuf: '0'
    type: float
    vlen: '1'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [640, 236.0]
[... 184 lines cut ...]
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [976, 108.0]
    rotation: 0
    state: enabled

connections:
- [blocks_add_const_vxx_0_0, '0', blocks_float_to_uchar_0, '0']
- [blocks_complex_to_float_0, '0', blocks_interleave_0, '0']
- [blocks_complex_to_float_0, '1', blocks_interleave_0, '1']
- [blocks_float_to_complex_0, '0', blocks_rotator_cc_0, '0']
- [blocks_float_to_uchar_0, '0', blocks_file_sink_0, '0']
- [blocks_interleave_0, '0', blocks_add_const_vxx_0_0, '0']
- [blocks_multiply_const_vxx_0, '0', rational_resampler_xxx_0, '0']
- [blocks_rotator_cc_0, '0', blocks_multiply_const_vxx_0, '0']
- [blocks_wavfile_source_0, '0', blocks_float_to_complex_0, '0']
- [blocks_wavfile_source_0, '1', blocks_float_to_complex_0, '1']
- [rational_resampler_xxx_0, '0', rational_resampler_xxx_1, '0']
- [rational_resampler_xxx_1, '0', blocks_complex_to_float_0, '0']

metadata:
  file_format: 1
```

argilo, 2020-08-04:
By the way, I put the work I've done so far into a pull request: https://github.com/theori-io/nrsc5/pull/232

pclov3r, 2020-08-05:
Might want to give this a try with my RSP DX. With that said I want to use it's HDR mode but it's lowest sample rate and I believe only is 2 MSPS using it.

Not quite sure how to reconfigure the re-sampler to down sample 2,000,000 down to 1,488,375. Any pointers? :) 

Again, My DSP knowledge is horrible and I need to learn but this isn't exactly the place for it.

argilo, 2020-08-05:
To resample from 2,000,000 to 1,488,375 just change the `samp_rate` variable from 500,000 to 2,000,000 and the Interpolation parameter in the second Rational Resampler block from 324 to 81.

The Phase Increment parameter in the Rotator block also needs to be updated to shift the signal up or down by the right amount.

pclov3r, 2020-08-05:
Thanks! I'll give it a try when conditions improve a bit and see if i can get anything.

pclov3r, 2020-08-05:
Bah keeps trying sync and drops out. Might improve at night will post if I do. 

Would you be so kind to explain the interpolation values for having a 8 MSPS I/Q file down to 1,488,375 samples? I would much appreciate it : )

pclov3r, 2020-08-05:
So, I was able to get to sync but it seems to have died rather quickly. Also throws a Unknown HDC block type:6 error so could be the cause. 

```05:10:29 Block @ 20
05:10:29 Sync!
05:10:29 ignoring partial pdu
Unknown HDC block type: 6```

Any ideas or is the signal too weak?

pclov3r, 2020-08-06:
I figured how how to make 8 MSPS work on my own. I do need to learn DSP stuff :)

Anyhow @argilo would you like me still to upload the HD AM capture to see if your able to do more with it or to get your judgement that it's too weak to decode?

argilo, 2020-08-06:
Yep, I would definitely appreciate samples, especially in cases where your hardware receiver is able to successfully decode.

pclov3r, 2020-08-06:
https://gofile.io/d/kjujxU 

Let me know your opinions of that capture. Centered on 1125 kHz. The signal in question is at 1140 kHz and this is KHTK via skip.

argilo, 2020-08-06:
It looks good! I was actually able to get audio out of it once I changed libfaad to accept HDC block type 6 as if it was type 1. There's some choppiness to the audio that makes me think there's something different about type 6, but at least it's intelligible!

I've pushed updated code in the pull request.

argilo, 2020-08-06:
By the way, if you're using my flow graph to convert to a .cu8 file, you can increase the constant in the "Multiply Const" block to 7000, since the signal level is lower in your capture than in @billybobgot's.

pclov3r, 2020-08-06:
Hum, Unable to get it sync with the same file. I made the GNU radio flow-graph changes. Seems it keeps finding blocks but wont sync. 

Any suggestions?

argilo, 2020-08-06:
Did you change the Rotator's phase increment to "-15000 * 2 * math.pi / samp_rate"?

argilo, 2020-08-06:
The samp_rate variable should be 2000000, multiply const should be 7000, and the rational resamplers should be 147/128 and 81/125. Any luck?

pclov3r, 2020-08-06:
I completely re-did the flow graph and it works fine now. I guess i missed something that wasn't obvious at first. 

Thanks!

pclov3r, 2020-08-06:
Just to add not much luck trying to use the SDRplay rtl_tcp compatible server to stream it in real time. Seems to only work if i use HDR mode with a I/Q recording and convert it later. Might work with a stronger signal.

The signal is already on the digital cliff so likely the issue.

PR #232 (2020-08-11): Add support for AM
Resolves #33.

This adds basic support for receiving AM signals, as described in [NRSC-5 Layer 1 AM](https://www.nrscstandards.org/standards-and-guidelines/documents/standards/nrsc-5-d/reference-docs/1012s.pdf).

To use the new functionality, simply add the `--am` command line option.

Some limitations:

* Only the hybrid mode (MA1) is supported. Fortunately, almost all stations (except WWFD) are using this mode.
* Audio quality is sub-optimal, because only the "core" audio stream (20 kbps) is decoded. I don't yet know how to decode the packets in the "enhanced" audio stream [... cut]
closed 2020-08-11 by argilo