theori-io/nrsc5 #36: Frequency offset(s) for live and sampled file playback?

erikson1970, 2017-07-28:
It looks like your sample output IQ baseband at 1.4MHz sample rate.  Further, it seems like you're saving the full IQ baseband.  It would be pretty slick if you could tune the receiver to an offset frequency of the baseband local oscillator freq when either playing back a sample from an IQ sample file or live from an RTL-SDR. Some RTLs have a big DC spike at the LO center freq which can contribute to poor SNR.  One further benefit is that it would allow you to tune to a different FM station on playback if the second station's full channel is in the IQ baseband sample window (ie within 1.4Mhz / 2.0 of the LO frequency)

I confess that I haven't dug through much of your code, so if you're doing this already, then kudos are due.

argilo, 2017-07-28:
> Some RTLs have a big DC spike at the LO center freq which can contribute to poor SNR.

Currently that doesn't matter, because the digital sidebands are well away from DC. The analog FM signal is in the centre, and it's filtered out because it's not needed. But if support for all-digital stations is ever added, then the DC spike (and 1/f noise present in some RTL-SDR dongles) would become a problem and offset tuning would be useful. (If anyone has a local digital-only station, please send us a recording!)

That said, being able to tune multiple stations from a single IQ file would be a handy feature and wouldn't be difficult to add, so I think it's a worthy feature request.

erikson1970, 2017-07-29:
The other thought/feature-request to stick on the table in the same general topic is to allow saving (to file) the baseband after down-sampling or decimating to the FM channel bandwidth of 200ish kHz.  The sample file sizes would shrink precipitously. I forked the project.  I will take a look at these.

argilo, 2017-07-29:
The bandwidth of an HD Radio channel is actually 400 kHz, but nrsc5 decimates to 744 kHz as that happens to be a convenient sample rate for OFDM demodulation. So it would be possible to cut the size of the recording in half, but it would be difficult to do more than that.

argilo, 2017-07-29:
Also, note that the 1448 ksps signal from the RTL-SDR is 8-bit while the decimated 744 ksps signal is 16-bit. So storing it as-is wouldn't save space. It would need to be truncated to 8 bits.