theori-io/nrsc5 #38: SIS messages?

itdaniher, 2017-08-01:
I'm interested in trying to use the SIS Parameter Message and possibly station location from the SIS PDU Messages to establish a crude time / location for a distributed sensing project, but it looks like this information isn't decoded right now. It looks like I'd need to start with `frame.c` - maybe in `frame_process` but I'd appreciate any pointers in the interim.

Cheers!

argilo, 2017-08-01:
@itdaniher SIS message type 4 would need to be implemented here:

https://github.com/theori-io/nrsc5/blob/12d9526f89838b38b14b4d25793c0ceb06e9b68d/src/pids.c#L122-L124

Note that SIS parsing is only on the experimental branch currently; it hasn't been merged to master yet.

The format is described in http://www.nrscstandards.org/SG/NRSC-5-C/1020sI.pdf section 4.4.

itdaniher, 2017-08-05:
Brilliant, thanks! I'm actually starting with processing time transmissions (MSG ID 0b0111 / first few indexes) / SIS Parameter Messages first, but it looks like doing so will require introducing state in the PIDS parsing to accumulate time metadata. I haven't looked enough at the code yet to figure out how to most elegantly do so - maybe define relevant structs in pids.h?

Cheers!

argilo, 2017-08-05:
Yep, storing state in a structs would make sense. There are other places that's done already.

argilo, 2017-09-30:
#79 implements most of the remaining pieces, apart from Universal Short Station Name.

argilo, 2019-03-02:
I'm going to close this off now, since nearly all SIS messages are now decoded.

closed 2019-03-02 by argilo