theori-io/nrsc5 #40: PTHREAD_SETNAME_NP and windows

pclov3r, 2017-08-02:
Sorry in advanced if am doing this incorrectly since i'm still new to github.

As for https://github.com/theori-io/nrsc5/commit/1378c991b873be778976d1bbf9cff5a7096e13c2 PTHREAD_SETNAME_NP exists at least with new versions of mingw-w64 to cross compile for windows. See https://github.com/msys2-contrib/mingw-w64/commit/0d95c795b44b76e1b60dfc119fd93cfd0cb35816

pclov3r, 2017-08-02:
Also for https://github.com/theori-io/nrsc5/commit/1378c991b873be778976d1bbf9cff5a7096e13c2 i'm still getting undefined reference to `CMPLXF'

Perhaps it's MinGW.

CMAKE output

```
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/i686-w64-mingw32-gcc
-- Check for working C compiler: /usr/bin/i686-w64-mingw32-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for strndup
-- Looking for strndup - not found
-- Looking for CMPLXF
-- Looking for CMPLXF - not found
-- Looking for __builtin_complex
-- Looking for __builtin_complex - not found
-- Looking for _Imaginary_I
-- Looking for _Imaginary_I - not found
-- Looking for pthread_setname_np in pthread
-- Looking for pthread_setname_np in pthread - found
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/nrsc5_branch_experimental_windows/build
```

pclov3r, 2017-08-02:
Also, Just to follow up src/CMakeLists.txt was modified so it could find pthread_setname_np

strndup appears to work tho.

awesie, 2017-08-02:
I'm working through a complete build using MSYS2 / MinGW. This is one of the more frustrating environments to have to work with. Once it is building and not crashing, I'll commit another fix.

pclov3r, 2017-08-02:
I've been using mingw-w64 on Debian 9 to cross compile it for Windows and it seems to be stable. Figure it would be the same for most Linux distributions.

I've not tried to use MYSYS2.

awesie, 2017-08-02:
1d19244 fixes compilation on Windows. You should be able to compile using MSYS2 and MinGW64.

pclov3r, 2017-08-02:
Yep it works fine now without having to define CMPLXF or strdup

As for the CMakeList modifications that need changed for faad and id3v2lib could be build environment specific so figure those will be left out.

Same goes for PTHREAD_SETNAME_NP I figure as well.

I'll go ahead and close this.

closed 2017-08-02 by pclov3r