theori-io/nrsc5 #43: Limit PIDS message output?

pclov3r, 2017-08-04:
This is a feature suggestion and perhaps an enhancement. Right now the FCC Facility ID and Call sign scroll by quickly making it difficult to read the other information. I figure it's decoded in real time as soon as it comes in.

Looks like it's decoded multiple times every second. 

Perhaps there is a way to limit the rate that it decided perhaps making it easier to read the other information such as the song artist and title?

argilo, 2017-08-05:
I've been thinking about that problem too. The program could really use a nice UI, but in the meantime it probably makes sense to slow down the flood of debug messages. Maybe it would make sense to display PIDS messages (and artist/title for that matter) only when they differ from what was received previously.

itdaniher, 2017-08-05:
This seems like it would need saved state the same as #38, interested to see how this is implemented!

pclov3r, 2017-08-05:
A UI would be nice but that sounds like alot of extra work.

For now i was thinking if you could somehow limit the PIDS output to like 1 message per second may help make stuff more visible.

argilo, 2017-09-17:
Solved by #57 (on the experimental branch for now).

PR #57 (2017-09-17): Only log SIS messages when there's a change.
Fixes #43.

Currently SIS logging is very noisy. To solve this, I'm proposing to log only when a PDU contains new values.

/cc @pclov3r @itdaniher
closed 2017-09-17 by argilo