theori-io/nrsc5 #45: Cygwin x86 (32 bit) compilation fails

K2DLS, 2017-08-05:
Compilation is failing because of some issue with CMPLXF.  I have verified that complex.h exists in the Cygwin environment.

[ 50%] Building C object src/CMakeFiles/nrsc5.dir/acquire.c.o
In file included from /home/dan/git/nrsc5/src/acquire.c:20:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 52%] Building C object src/CMakeFiles/nrsc5.dir/decode.c.o
In file included from /home/dan/git/nrsc5/src/decode.h:4:0,
                 from /home/dan/git/nrsc5/src/decode.c:19:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 55%] Building C object src/CMakeFiles/nrsc5.dir/frame.c.o
In file included from /home/dan/git/nrsc5/src/frame.c:18:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 58%] Building C object src/CMakeFiles/nrsc5.dir/hdc_to_aac.c.o
[ 61%] Building C object src/CMakeFiles/nrsc5.dir/input.c.o
In file included from /home/dan/git/nrsc5/src/input.c:20:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 64%] Building C object src/CMakeFiles/nrsc5.dir/main.c.o
In file included from /home/dan/git/nrsc5/src/main.c:32:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 67%] Building C object src/CMakeFiles/nrsc5.dir/output.c.o
In file included from /home/dan/git/nrsc5/src/output.c:23:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 70%] Building C object src/CMakeFiles/nrsc5.dir/pids.c.o
In file included from /home/dan/git/nrsc5/src/pids.c:18:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 73%] Building C object src/CMakeFiles/nrsc5.dir/sync.c.o
In file included from /home/dan/git/nrsc5/src/sync.c:18:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 76%] Building C object src/CMakeFiles/nrsc5.dir/firdecim_q15.c.o
In file included from /home/dan/git/nrsc5/src/firdecim_q15.h:3:0,
                 from /home/dan/git/nrsc5/src/firdecim_q15.c:12:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 79%] Building C object src/CMakeFiles/nrsc5.dir/firdes_kaiser.c.o
[ 82%] Building C object src/CMakeFiles/nrsc5.dir/resamp_q15.c.o
In file included from /home/dan/git/nrsc5/src/resamp_q15.h:3:0,
                 from /home/dan/git/nrsc5/src/resamp_q15.c:12:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 85%] Building C object src/CMakeFiles/nrsc5.dir/math.c.o
In file included from /home/dan/git/nrsc5/src/math.c:1:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
[ 88%] Building C object src/CMakeFiles/nrsc5.dir/conv_dec.c.o
In file included from /home/dan/git/nrsc5/src/conv_dec.c:30:0:
/home/dan/git/nrsc5/src/defines.h: In function ‘cq15_to_cf’:
/home/dan/git/nrsc5/src/defines.h:55:12: warning: implicit declaration of function ‘CMPLXF’ [-Wimplicit-function-declaration]
     return CMPLXF((float)cq15.r / 32767.0f, (float)cq15.i / 32767.0f);
            ^
In file included from /home/dan/git/nrsc5/src/conv_dec.c:38:0:
/home/dan/git/nrsc5/src/conv_dec.c: At top level:
/home/dan/git/nrsc5/src/conv_gen.h:129:13: warning: ‘gen_metrics_k5_n2’ defined but not used [-Wunused-function]
 static void gen_metrics_k5_n2(const int8_t *seq, const int16_t *out,
             ^
/home/dan/git/nrsc5/src/conv_gen.h:138:13: warning: ‘gen_metrics_k5_n3’ defined but not used [-Wunused-function]
 static void gen_metrics_k5_n3(const int8_t *seq, const int16_t *out,
             ^
/home/dan/git/nrsc5/src/conv_gen.h:148:13: warning: ‘gen_metrics_k5_n4’ defined but not used [-Wunused-function]
 static void gen_metrics_k5_n4(const int8_t *seq, const int16_t *out,
             ^
/home/dan/git/nrsc5/src/conv_gen.h:159:13: warning: ‘gen_metrics_k7_n2’ defined but not used [-Wunused-function]
 static void gen_metrics_k7_n2(const int8_t *seq, const int16_t *out,
             ^
/home/dan/git/nrsc5/src/conv_gen.h:179:13: warning: ‘gen_metrics_k7_n4’ defined but not used [-Wunused-function]
 static void gen_metrics_k7_n4(const int8_t *seq, const int16_t *out,
             ^
[ 91%] Building C object src/CMakeFiles/nrsc5.dir/reed-solomon.c.o
[ 94%] Building C object src/CMakeFiles/nrsc5.dir/galois.c.o
[ 97%] Building C object src/CMakeFiles/nrsc5.dir/log.c.o
[100%] Linking C executable nrsc5.exe
CMakeFiles/nrsc5.dir/input.c.o:input.c:(.text+0x49e): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/sync.c.o:sync.c:(.text+0xb6): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/sync.c.o:sync.c:(.text+0x7ef): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/sync.c.o:sync.c:(.text+0x84e): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/resamp_q15.c.o:resamp_q15.c:(.text+0x62b): undefined reference to `CMPLXF'
CMakeFiles/nrsc5.dir/resamp_q15.c.o:resamp_q15.c:(.text+0xd1d): more undefined references to `CMPLXF' follow
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/nrsc5.dir/build.make:516: src/nrsc5.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:161: src/CMakeFiles/nrsc5.dir/all] Error 2
make: *** [Makefile:128: all] Error 2

awesie, 2017-08-05:
I assume you are using the experimental branch. Can you remove your build directory, and post the messages you get when you run cmake? It should look something like the following:

```
$ cmake -G "MSYS Makefiles" ..
-- The C compiler identification is GNU 7.1.0
-- Check for working C compiler: C:/msys64/mingw64/bin/gcc.exe
-- Check for working C compiler: C:/msys64/mingw64/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for strndup
-- Looking for strndup - not found
-- Looking for CMPLXF
-- Looking for CMPLXF - not found
-- Looking for _Imaginary_I
-- Looking for _Imaginary_I - not found
-- Looking for _Complex_I
-- Looking for _Complex_I - found
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/awesie/nrsc5/build
```

K2DLS, 2017-08-05:
OK, experimental branch!  Up and running now under Cygwin 32 bit environment on Windows 10.  Are there any performance enhancements that will take advantage of a 64 bit environment?  If so I will try next on Cygwin x86-64.  Please advise.

Thanks for all the good work on this!

closed 2017-08-05 by K2DLS

awesie, 2017-08-05:
There should not be major benefits to 64-bit. You can try building 32-bit with: cmake -DUSE_SSE=ON, if you want better performance.

K2DLS, 2017-08-05:
Will do.
Should the -l switch function in the experimental branch?  It seems to just turn off logging if I supply a numeric value, ie -l 6.

awesie, 2017-08-05:
The -l switch chooses the logging level, with -l 0 as the most logging and -l 5 as the least logging.