theori-io/nrsc5 #47: Basic samp_rate & data encoding format for use w/GNU Radio

Teque5, 2017-08-08:
I've tried piping my own data into nrsc5 and I've also tried reading in your sample file. I could futz with it for another hour, but it would be useful if you told us what the sample rate and encoding was of the sample file in the /support directory.

My current <incorrect> guess (since I can't demod it normally):
samp_rate = 1488375
encoding = interleaved signed short (16 bit I, 16 bit Q)

When I use your tool the file is around 9 seconds long, which at 1448KHz sample rate would be about 4 bytes per sample.

What do I have wrong? Is there a header in the sample file?

awesie, 2017-08-08:
It is designed for rtl-sdr, so the input is what I get from librtlsdr. I believe it is interleaved unsigned byte. The sample rate should be 1488375.

See https://www.reddit.com/r/RTLSDR/comments/2qrfvn/help_with_iq_samples/cn8w01e/.

argilo, 2017-08-08:
The sample rate is 1488375 and the format is interleaved unsigned bytes. The same format that rtl_sdr uses.

Teque5, 2017-08-08:
Ok. So the file type is complex interleaved 8 bit. The following GNU Radio flowgraph may help someone else. The screenshot also shows how to convert USRP complex float into this format.
[image]

closed 2017-08-08 by Teque5

argilo, 2017-08-08:
@Teque5 That doesn't look right. You would need to add / subtract 127 when converting signed to/from unsigned bytes.

Teque5, 2017-08-08:
@argilo The interleaved short to complex block (and vice-versa) has this built-in within GNU Radio.

argilo, 2017-08-08:
Yep, that looks like the right way.