theori-io/nrsc5 #49: Play HD Radio Throughout Your Home Using Google Cast or Airplay

classicjazz, 2017-08-17:
I found that the nrsc5 app was perfect as an audio source for multicasting an HD Radio station throughout my home using Google Cast or Airplay.

I wrote a brief article about it here:
http://freetime.mikeconnelly.com/archives/5052

If anyone has any recommended updates or corrections, I would appreciate them.

Great work!

tylert, 2017-08-21:
@classicjazz Nice!  Sorry to piggy-back on your issue but I also figured it would be really nice to have some form of streaming functionality built right into nrsc5 itself without requiring piping the output to something so it'll work better with other things like VLC and so on too.  Say, stream all programs 0, 1, 2, etc. to different ports with a specified offset, e.g.:  9000, 9001, 9002, etc.

I envision an RTL-SDR and a small computer with nrsc5 acting as a network-attached HD Radio receiver placed somewhere convenient for best reception.

FYI @argilo :)

classicjazz, 2017-08-21:
I updated my article with a simple BASH script that simplifies using nrsc with SOX down to 2 parameters (frequency and band) and will autofill these if not given. I prefer using either SOX or MPV so as to chain audio filters that compensate for the companding and compression artifacts with HD Radio.

I am still giving some thought to a simple selector of favorite radio stations.

classicjazz, 2017-08-21:
@tylert 
If you want to stream an individual stream (not all bands like you wish), you can do that now by piping the output from NRSC to SOX to NETCAT. See https://medium.com/@luigifcruz/listening-live-fm-radio-over-the-network-with-rtl-sdr-2349c5abe878 for an example.

If you don't want to leave a Mac running with Airfoil or if that isn't proximate to your antenna, then you could use a RPi3 to capture the steam and pipe it to an internet feed that you tune with a browser. The browser page could then serve that up for casting to internet clients. See http://radio.mikeconnelly.com for example. 

Although those are internet streams, you could do something similar with a LAN based feed, obtaining the ID3 tags and album art from the HD Radio feed.

classicjazz, 2017-09-06:
I updated the article to reflect the simpler frequency parameter (e.g. "90.9") supported in the latest build. Also, I rewrote the instructions and bash script using MPV.

classicjazz, 2017-10-28:
I updated the article including recommending WAV output over ADTS to avoid some audio players' issues with AAC SBR extensions.

closed 2020-04-03 by classicjazz