theori-io/nrsc5 #50: Permit frequency shorthand of "M" (e.g. 90.9M instead of 90900000)

classicjazz, 2017-08-20:
In rtl_fm, you can specify a frequency of 90.9M instead of writing out all the zeros like 90900000.

In nrsc5, if you specify a frequency of 90.9M, it will do auto gain control but not play anything. 

I have tried a number of stations with the same result. Unless I am missing something...

argilo, 2017-08-21:
nrsc5 doesn't support that, but I think it would be a useful feature. I'll put it in the next time I'm working on the project.

classicjazz, 2017-08-22:
Thank you.

Some additional testing...

If I ask RTL_FM to tune 90.1M, it consistently tunes 90371000 Hz, not 90100000 Hz. 90100000 Hz works fine for nrsc5 but  90371000 Hz results in 

`Tuning to the radio station: 90.3 Mhz HD1
16:36:56 INFO  main.c:183: [0] Generic RTL2832U OEM
Playing: -
[file] Reading from stdin...
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
16:36:57 INFO  main.c:63: Gain: 0.0 dB, CNR: -16.127131 dB
16:36:57 INFO  main.c:63: Gain: 0.9 dB, CNR: -18.021013 dB
16:36:57 INFO  main.c:63: Gain: 1.4 dB, CNR: -18.603493 dB
16:36:57 INFO  main.c:63: Gain: 2.7 dB, CNR: -19.879759 dB
16:36:57 DEBUG main.c:67: Best gain: 0`

Similarly with 90.9Mhz HD1 or 909000000 specified:
In RTL_FM, it tunes to 91171000 Hz instead.

argilo, 2017-08-22:
I suspect that's just because rtl_fm is using offset tuning to keep the signal of interest away from the noise spike around 0 Hz.

nrsc5 doesn't have that problem because the noise spike is in the analog FM signal, which is filtered out anyway.

classicjazz, 2017-08-22:
Ah, makes sense. I was confused because I was looking for the exact frequency to plug in. And what worked in analog wasn't working in digital. 

All the more reason to just simplify the UX and just use 90.9M or similar...

awesie, 2017-09-05:
This is fixed in #54. Instead of a prefix, it will accept a decimal number for the frequency, and if it is less than 10,000, it will be multiplied by 1M.

PR #54 (2017-09-05): Accept frequency in MHz or Hz.
Fixes #50.

This change allows the frequency to be specified either in MHz or Hz. If the number is greater than 10,000 it will be interpreted as Hz, otherwise as MHz. I chose that threshold because the RTL-SDR can't tune that high (in MHz) or that low (in Hz).

If the frequency is entered as `107.1M` (like `rtl_fm`) then the trailing `M` is simply ignored and it's interpreted as 107.1 MHz.
closed 2017-09-05 by awesie