theori-io/nrsc5 #51: NRSC stops after about 45 minutes

classicjazz, 2017-08-22:
I am finding that NRSC is ceasing to pipe audio to SOX PLAY/MPV after about 45-50min. The timer on SOX / MPV ceases to count upward.

Typical Mac Terminal output is as follows. I can create an ADTS dump file if necessary. In this case, the station is WHYY HD1. My exact settings are per the shell script described in Issue 49.

```
Tuning to the radio station: 90.1 Mhz HD1
Playing: -
[file] Reading from stdin...
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
 (+) Audio --aid=1 (pcm_s16le 2ch 44100Hz)
AO: [coreaudio] 48000Hz stereo 2ch float
Unexpected block type: 2, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 1, expected 6
Unexpected block type: 7, expected 6
Unexpected block type: 5, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 7, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 3, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 4, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 5, expected 6
Unexpected block type: 1, expected 6
Unexpected block type: 1, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 1, expected 6
Unexpected block type: 5, expected 6
Unexpected block type: 0, expected 6
Unexpected block type: 5, expected 6
Unexpected block type: 4, expected 6
Unexpected block type: 2, expected 6
^C 00:45:32 / 00:00:00
```

argilo, 2017-08-22:
I've observed this problem as well. I think it's because the symbol timing (`samperr`) drifts to the maximum or minimum value over time because the arbitrary resampler is not properly adjusted. Once that happens, decoding is no longer possible because sync is lost.

I'm of the mind to just get rid of the arbitrary resampler and drop/add samples as necessary to keep `samperr` centered.

argilo, 2017-08-22:
Another possible solution would be to have the arbitrary resampler's feedback loop try to keep samperr centered rather than trying to make its rate of change zero.

classicjazz, 2017-08-22:
Running it with full logging, dumped to out.dts, I see the following:

`15:47:32 DEBUG pids.c:114: Station Name: WRTI-FM
15:47:32 DEBUG pids.c:103: Country: US, FCC facility ID: 65190
15:47:33 INFO  sync.c:321: MER: nan dB (lower), nan dB (upper)
15:47:33 INFO  decode.c:77: BER: 0.062754, avg: 0.002589, min: 0.000027, max: 0.062754
15:47:33 DEBUG sync.c:222: lost sync (-1, -1)!
15:47:38 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:41 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:44 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:47 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:50 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:53 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:56 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:47:59 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:02 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:05 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:08 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:11 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:14 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:16 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:19 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:22 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:25 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:28 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:31 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:34 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:37 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:40 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:43 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:46 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:49 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:52 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000
15:48:55 DEBUG acquire.c:99: Timing offset: 0.000000, slope: 0.000000`

I killed the app because this just repeats forever.

[out.adts.zip](https://github.com/theori-io/nrsc5/files/1243191/out.adts.zip)

FYI:  I modified that bash script on my site to allow users to easily create an out.dts.

closed 2017-09-06 by awesie

classicjazz, 2017-09-07:
On a strong station (WHYY), I just ran nrsc5 for 13.5 hrs nonstop. Thank you!

argilo, 2017-09-07:
@classicjazz I'm happy to hear it's fixed. Thanks for testing!

PR #56 (2017-09-06): Fix cfo_idx integer overflow.
Fixes #51 (and hopefully also #53).

I tracked down the reason the receiver stops after 48 minutes. It's because `cfo_idx` (a 32-bit signed integer) overflows, wrapping around to a negative number, at which point `st->cfo_idx++ % FFT` becomes negative, causing out-of-bounds accesses to `st->cfo_tbl` here:

https://github.com/theori-io/nrsc5/blob/b9acf8eb439e8a1ba783065aa9dd7c991d5bac12/src/input.c#L97

I've solved this by reducing `cfo_idx` mod `FFT` after the loop.

/cc @classicjazz