theori-io/nrsc5 #53: ADTS decoding frequently throws errors in MPV

classicjazz, 2017-08-24:
I was testing whether it was better to pipe ADTS v WAV to MPV. If I understand correctly, the ADTS stream would be an (essentially) unmodified copy of the AAC audio feed from the station. WAV output requires NRSC to decode those packets to PCM.

In theory, ADTS would be better because all audio processing would be done within MPV and less data is piped between the programs. But, if I run NRSC5 long enough, I frequently see ADTS decoding errors like what follows. So, WAV output seems to be more stable (or at least whatever errors there are in the ADTS stream are accommodated by NRSC in when it transcodes to WAV before piping to MPV). 

Eventually, the program halts. I don't know whether this is related to Issue 51 or a separate issue specific to ADTS decoding.

The exact code that I am using is:
**nrsc5 -o - -q -f adts $FREQUENCY $BAND | mpv - --audio-samplerate=48000 -af lavfi='extrastereo'**

```
Tuning to the radio station: 88.5 Mhz HD1
Playing: -
[file] Reading from stdin...
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
 (+) Audio --aid=1 (aac 2ch 44100Hz)
AO: [coreaudio] 48000Hz stereo 2ch float
A: 00:48:01 / 00:00:00
[ffmpeg/audio] aac: Number of bands (48) exceeds limit (45).
Error decoding audio.
A: 01:35:56 / 00:00:00
[ffmpeg/audio] aac: Reserved SBR extensions is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
[ffmpeg/audio] aac: If you want to help, upload a sample of this file to ftp://upload.ffmpeg.org/incoming/ and contact the ffmpeg-devel mailing list. (ffmpeg-devel@ffmpeg.org)
[ffmpeg/audio] aac: Expected to read 14 SBR bytes actually read 47.
[ffmpeg/audio] aac: Multiple frames in a packet.
[ffmpeg/audio] aac: Reserved bit set.
[ffmpeg/audio] aac: Number of bands (19) exceeds limit (15).
Error decoding audio.
A: 02:23:50 / 00:00:00
```

parkerlreed, 2017-09-01:
At 39 minutes I had a couple blips in audio. Still playing.

parkerlreed, 2017-09-01:
Died at the 47 minute mark

````
[parker@yoga710 ~]$ nrsc5 -o - -q -f adts 89500000 1 | mpv - --audio-samplerate=48000 -af lavfi='extrastereo'
Playing: -
[file] Reading from stdin...
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
 (+) Audio --aid=1 (aac 2ch 44100Hz)
AO: [pulse] 48000Hz stereo 2ch float
(Buffering) A: 00:47:49 / 00:00:00 Cache:  0s+4KB
````

This is in Linux

argilo, 2017-09-06:
@classicjazz @parkerlreed Is this working better now that #51 is fixed?

parkerlreed, 2017-09-06:
@argilo It'll be a little bit before I can test a 45 minute run :smile: 

Thanks for the update.

classicjazz, 2017-09-06:
On a strong signal (WHYY), I have tested 1hr 30min for ADTS output without stream failure or FFMPEG errors.

We have a rainy day here and I still need to test a weaker signal (WXPN) that throws a number of unexpected block errors.

argilo, 2017-09-06:
I tested this on a weak station today and twice nrsc5 failed with:

```
nrsc5: /home/argilo/git/nrsc5/src/bitreader.h:26: br_read1bit: Assertion `br->buf != br->end' failed.
```

So it looks like ADTS output still has problems. I'll leave this issue open for the time being.

argilo, 2017-09-06:
I think the problem is that the HDC-to-AAC converter is operating under the assumption that the HDC input is valid. But NRSC-5 only uses an 8-bit CRC for audio packets, so there's a 1 in 256 chance that an audio packet containing errors will be accepted. When the signal is weak, it doesn't take long for that situation to occur.

I'll see if I can beef it up to accept any input, valid or invalid.

argilo, 2017-09-07:
It turns out the problem is not limited to the adts output mode. I made an elementary fuzzer for HDC (which just flips bits from valid HDC packets at random) and when I feed its output into nrsc5 it readily crashes. It looks like a bunch of bounds checks will need to be added to the HDC decoders in nrsc5.

classicjazz, 2017-09-07:
Thanks, @argilo.

Separately, answering my question above, I think this commit explains that nsrc5 is not transcoding (lossy reencoding) the HDC-format audio stream. But simply correcting the proprietary HDC format as compliant AAC and then reencapuslating the AAC within an MPEG2 TS container format. 

https://github.com/theori-io/nrsc5/commit/a4e4327b577cf0282fd3d3a5eaec71d9ad6e9087

argilo, 2017-09-07:
Yes, that's correct. The conversion from HDC to AAC is lossless because it just re-orders the data.

classicjazz, 2017-10-19:
As requested, attached are three audio captures and accompanying log files for WHYY, WRTI, and WXPN.

As mentioned in issue https://github.com/theori-io/nrsc5/pull/82, all three broadcast from the same antenna farm about 15 miles away. The difference is that WHYY is 13500 watts ERP, WRTI is 7700 ERP, and WXPN is 2650 watts ERP. Further, WXPN is only 105 watts digital.

[whyy.txt](https://github.com/theori-io/nrsc5/files/1398824/whyy.txt)

[wrti.txt](https://github.com/theori-io/nrsc5/files/1398825/wrti.txt)

[wxpn.txt](https://github.com/theori-io/nrsc5/files/1398826/wxpn.txt)

classicjazz, 2017-10-19:
> If you are listening to stream that uses the DRM PS extension with ADTS output, you may get the following output repeatedly:
> [ffmpeg/audio] aac: Reserved SBR extensions is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
> As the error message explains, this is due to most versions of FFmpeg not supporting DRM PS with AAC-SBR.

I am using 
ffmpeg version 3.4 Copyright (c) 2000-2017 the FFmpeg developers
  built with Apple LLVM version 9.0.0 (clang-900.0.38)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/3.4 --enable-shared --enable-pthreads --enable-version3 --enable-hardcoded-tables --enable-avresample --cc=clang --host-cflags='-I/Library/Java/JavaVirtualMachines/jdk-9.jdk/Contents/Home/include -I/Library/Java/JavaVirtualMachines/jdk-9.jdk/Contents/Home/include/darwin' --host-ldflags= --enable-gpl --enable-libbluray --enable-libfdk-aac --enable-libmp3lame --enable-libx264 --enable-libx265 --enable-libxvid --enable-opencl --enable-videotoolbox --disable-lzma --enable-nonfree

classicjazz, 2017-10-19:
Here's another WHYY capture. I haven't changed any settings on the antenna (amplifier or placement). Usually, WXPN is very hard to tune and WHYY is easy. The opposite seems true today.

[whyy 2.adts.zip](https://github.com/theori-io/nrsc5/files/1399729/whyy.2.adts.zip)
[whyy 2.txt](https://github.com/theori-io/nrsc5/files/1399730/whyy.2.txt)

srsampson, 2018-07-05:
bitreader.h line 32 is:     `return !!(br->byte & 0x100);`

The double bang seems suspicious. Maybe a comment on what it is supposed to do??

awesie, 2019-02-12:
With the latest branch, ADTS transcoding is deprecated. The only supported output format is WAV and HDC, though that is probably useless for anyone except developers. Closing.

closed 2019-02-12 by awesie