theori-io/nrsc5 #61: Extracting IDv3 info from NRSC5 and passing to MPV

classicjazz, 2017-09-07:
I see that NRSC can see INFO tags including Title and Artist.

I also see @argilo 's commits for ID3v2
https://github.com/theori-io/nrsc5/commit/320784dca2233ade3660ac6a1c4bd799761324dc

Is it possible to extract title, artist and album art and pass it through a pipe to a player like MPV?

This would seem a cleaner approach than the more verbose logging mode (-l 2) in NRSC.

argilo, 2017-09-07:
I've been thinking about this a bit too. It would be great to pass ID3 tags through to the ADTS output.

KeyserSoze1, 2017-09-08:
I appreciate the credit but there wasn't much to what I did.  @argilo deserves a lot more for the effort he has put into this as it's pretty impressive how far it has gotten.  

I'm only semi-familiar with the ADTS format but I think you can embed the ID3 tags in the aac data.  However I'm not sure how many players support it.  I wonder if you wrote an ADTS frame with just ID3 tags how players would handle it?  That way you could just interleave ID3 tags periodically in the stream (possibly only when the ID3 tag changes).

As far as piping data into mpv, I'm not aware of any existing mechanism to do that.  Mpv generally receives metadata from the file itself.

classicjazz, 2017-09-08:
Since WAV doesn't support metadata, would FLAC output be a better alternative to ADTS and WAV?

awesie, 2019-02-12:
I had looked into this a bit a while ago. The short answer is that there doesn't seem to be wide support for inline tags (like artist, title, etc.) embedded in an open source container format. Most players seem to expect these tags to never change once set in a stream.

If someone has a counter example, I would be interested in knowing more. My conclusion thus far is that no off-the-shelf player will provide the support needed for this issue.