theori-io/nrsc5 #68: Pop / click suppression

classicjazz, 2017-09-22:
I have noticed that when a signal is low, there may be audible pops / clicks / or 1 second "whistles" in the feed from NRSC to MPV.

This seems to coincide with these log notices:
WARN  frame.c:511: crc mismatch!

Is there a way that we could suppress the pops with either 
1) worst case, silence or
2) preferably, an interpolation of the missing data (analog wave)

I have unsuccessfully searched for FFMPEG / MPV filters similar to Audacity's pop/click removal to solve this in real time on the audio stream. 

But it really would be better to have this functionality in NRSC because it can be correlated with the CRC mismatches.

pclov3r, 2020-06-29:
Seems like this is still kinda an issue? This however has been open for quite some time.

argilo, 2025-05-05:
Closing, as this should have been resolved by #399.

closed 2025-05-05 by argilo