theori-io/nrsc5 #73: Switch program streams on the fly

pclov3r, 2017-09-24:
So i'm curious if it's possible to switch between the different programs a station may have without having quit the program in again and going though the entire process again.

I think this also goes back to one of my earlier issues in which you said this needs a GUI. 

Perhaps this is too much work or not possible. 

What do you think?

argilo, 2017-09-24:
It's definitely a feature worth having, so let's leave this issue open. Even without a GUI, it should be possible to accept keyboard input to switch streams. It would also be nice for nrsc5 to tell you which programs are present.

pclov3r, 2017-09-24:
Perhaps something among the lines of something line of a 0-3 numeric input would switch the stream so you wouldn't have to go tough the entire process having to re-acquire the channel again etc.

KeyserSoze1, 2017-10-28:
While mine was strictly just hacking something together for a demo this should be very easily done.  I was able to send a udp packet to nrsc5 to have it switch on the fly.  In my case I had gqrx send a udp packet to a specific port on localhost that nrsc5 was listening to.  There were no issues with this and it worked as expected.  If I have time I'll look at adding keyboard input for switching programs and push a pull request if @argillo doesn't get to it first.

pclov3r, 2019-03-03:
Figure you haven't had a chance to look into this. Still would like this feature :)

argilo, 2019-03-03:
This feature seems like it would be more appropriate for a GUI application where there's an easy way to accept user input. The nrsc5 API already supports this use case, since it reports data pertaining to all streams through the callback function. It's up to the application to decide which stream (or streams) it cares about.

Since we now have a Python API, I was thinking I might try switching @cmnybo's [nrsc5-gui](https://github.com/cmnybo/nrsc5-gui) to use it. Adding a feature to switch streams should be straightforward.

awesie, 2019-03-03:
I generally agree with @argilo. I put together a quick hack if we want to support it in the CLI, though I am fine with the answer being no.

closed 2019-03-09 by argilo

argilo, 2019-03-31:
It's now possible to switch streams during playback in NRSC5-GUI as well. My changes are available in this pull request: https://github.com/cmnybo/nrsc5-gui/pull/9

commit a40277e2 (2019-03-09): Add simple keyboard commands to CLI. (#197)

* Add simple keyboard commands to CLI.

Resolves #73.

* Use _getch on Windows.

* Use read() instead of getchar().

This seems to fix issues with pthread_cancel not interrupting getchar.

* Make CLI interactive only if stdin is a tty.

* Stop the radio thread if user wants to quit.