theori-io/nrsc5 #74: error for object 0x7f8d4280a600: incorrect checksum for freed object - object was probably modified after being freed.

classicjazz, 2017-09-24:
Using this morning's master...

> Tuning to the radio station: 90.9 Mhz HD1
10:39:28 INFO  main.c:189: [0] Generic RTL2832U OEM
Playing: -
[file] Reading from stdin...
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
10:39:29 INFO  main.c:56: Gain: 0.0 dB, Power: -42.830997 dBFS
10:39:29 INFO  main.c:56: Gain: 0.9 dB, Power: -40.005108 dBFS
10:39:29 INFO  main.c:56: Gain: 1.4 dB, Power: -38.784588 dBFS
10:39:29 INFO  main.c:56: Gain: 2.7 dB, Power: -35.047985 dBFS
10:39:29 INFO  main.c:56: Gain: 3.7 dB, Power: -33.255219 dBFS
10:39:29 INFO  main.c:56: Gain: 7.7 dB, Power: -30.284607 dBFS
10:39:29 INFO  main.c:56: Gain: 8.7 dB, Power: -28.886299 dBFS
10:39:29 INFO  main.c:56: Gain: 12.5 dB, Power: -27.129055 dBFS
10:39:29 INFO  main.c:56: Gain: 14.4 dB, Power: -25.824263 dBFS
10:39:29 INFO  main.c:56: Gain: 15.7 dB, Power: -24.120682 dBFS
10:39:29 INFO  main.c:56: Gain: 16.6 dB, Power: -22.714890 dBFS
10:39:29 INFO  main.c:56: Gain: 19.7 dB, Power: -19.854498 dBFS
10:39:29 INFO  main.c:56: Gain: 20.7 dB, Power: -18.455906 dBFS
10:39:29 INFO  main.c:56: Gain: 22.9 dB, Power: -15.732992 dBFS
10:39:29 INFO  main.c:56: Gain: 25.4 dB, Power: -14.250073 dBFS
10:39:29 INFO  main.c:56: Gain: 28.0 dB, Power: -11.687900 dBFS
10:39:29 INFO  main.c:56: Gain: 29.7 dB, Power: -11.212385 dBFS
10:39:29 INFO  main.c:56: Gain: 32.8 dB, Power: -10.266764 dBFS
10:39:30 INFO  main.c:56: Gain: 33.8 dB, Power: -10.013288 dBFS
10:39:30 INFO  main.c:56: Gain: 36.4 dB, Power: -9.602272 dBFS
10:39:39 INFO  sync.c:248: Synchronized!
10:39:40 INFO  sync.c:324: MER: -6.600716 dB (lower), -13.836862 dB (upper)
10:39:40 INFO  decode.c:77: BER: 0.181901, avg: 0.181901, min: 0.181901, max: 0.181901
nrsc5(69910,0x7000020ba000) malloc: *** error for object 0x7f8d4280a600: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
[ffmpeg/demuxer] wav: Cannot check for SPDIF
 (+) Audio --aid=1 (pcm_s16le 2ch 44100Hz)

Exiting... (Errors when loading file)

argilo, 2017-09-24:
@classicjazz Does this happen consistently, or did you get it just once? If you can reproduce, please see if you can make a recording (with the `-w` option).

classicjazz, 2017-09-24:
I have seen it only once so far. But I am having a number of issues with the new build playing audio. Often times, it jumps between syncronizing and synch'ed. But I don't hear anything. Then I stop and restart. And it plays ok, albeit with packet drops.

classicjazz, 2017-10-28:
I haven't seen this recently. So I will close it until it reoccurs and I have more info.

closed 2017-10-28 by classicjazz

argilo, 2017-10-28:
Sounds good. We've made some robustness improvements since this issue was opened, so the issue may already be solved.