theori-io/nrsc5 #80: Analog to Digital migration behavior to match a psychical HD radio?

pclov3r, 2017-09-30:
Now, I'm thinking this may be far too much work with little benefit but I'll throw the idea out anyway.

Is it possible to start with Analog FM and than once the digital channel has been acquired than switch to digital? 

Basically the goal of matching the behavior of how a physical HD radio works. 

Thanks!

argilo, 2017-10-01:
It probably would be a lot of work to add that feature. It's not something I'm particularly interested in, but I'll leave the issue open in case someone else feels like implementing it someday.

awesie, 2017-10-01:
The difficult part is figuring out how much delay we have processing the digital audio. And then combine this with the information in the PDU header to determine how much we need to delay the analog audio.

classicjazz, 2017-10-01:
Unlike a moving car, NSRC5 is likely played on a stationary computer. So, my  preference would be pure digital reception. But speeding up initial audio playback is desirable.

There's been a lot of progress made on AGC timing (reducing it from 3-5 seconds to 1 second). 

It seems like the next things to focus on are reducing the time to "Synchronized" (5-19 seconds) and then audio playback (3-4 seconds). See more anecdotal details here: https://github.com/theori-io/nrsc5/pull/82#issuecomment-333350394

argilo, 2017-10-01:
@classicjazz There is a certain amount of latency inherent in the NRSC-5 standard (primarily coming from the interleavers) and so the acquisition time will always be at least a few seconds. But I believe there is still quite a bit of room for improvement in nrsc5.

pclov3r, 2017-10-01:
@awesie @argilo I agree. It seems the focus should be put on getting it to acquire and sync faster if possible vs adding analog FM and waiting for it to sync and than switching.

classicjazz, 2017-10-07:
@argilo 

I timed tuning secondary channels on the HD radio in my car. It consistently takes 5 seconds to jump between stations' HD2 channels from the time it is selected until audio is played. This seems like a useful benchmark for the art of the possible.

classicfm, 2017-11-16:
How about an option to exit with an error if not synchronized after a certain time? This would allow applications using nrsc5 to failover to using RTL_FM or SoftFM.

I read somewhere that many stations don't have HD capability on their back up equipment, for cost reasons. These backup facilities get used more than you think, for maintenance or whatever, and so we have all become used to 24/7 FM always being there. With the migration behavior of physical radios, I'd bet most people don't even notice when HD is down.

pclov3r, 2017-11-20:
@classicfm that could work.

classicfm, 2017-11-22:
I just want to point this out in case blending is actually implemented: this is by far the best stereo FM app I have found, and being a patch buried on a mailing list it's not anywhere near as well known as it should be. 

http://article.gmane.org/gmane.comp.mobile.osmocom.sdr/299

I should also note that it supports the FM deemphasis value used in the U.S., many other FM apps (e.g. SoftFM) have hardcoded the 1/50 microsecond value used in the EU.