theori-io/nrsc5 #81: Auto Gain may not work for some stations.

pclov3r, 2017-09-30:
So I noticed that the NPR station for my area the auto-gain feature does not work. It will increase it to max gain and then fail to acquire the channel.

If I set the gain manually to say 20.7 dB it works fine.

I think this may be caused by the NPR station having very high FM deviation levels often. 

Here is a screenshot of the FFT. 
[image]

pclov3r, 2017-09-30:
It did use to work before with the old method so I think https://github.com/theori-io/nrsc5/pull/65 may have caused the issue.

It seems all other stations work fine with it.

argilo, 2017-09-30:
What gain value does the automatic gain selection choose for that station?

pclov3r, 2017-09-30:
It will choose the max gain of 49.6 dB.

argilo, 2017-09-30:
Can you show me screenshots of the FFT with the gain set to 20.7 and 49.6? Also, could you cut & paste the debug messages printed by the gain selection? Thanks.

pclov3r, 2017-09-30:
Here is a log output

> 16:17:33 INFO  main.c:192: [0] Generic RTL2832U OEM
> 16:17:33 INFO  main.c:192: [1] Generic RTL2832U OEM
> Found Rafael Micro R820T tuner
> Exact sample rate is: 1488375.071248 Hz
> 16:17:34 INFO  main.c:56: Gain: 0.0 dB, Power: -44.875027 dBFS
> 16:17:34 INFO  main.c:56: Gain: 0.9 dB, Power: -42.619064 dBFS
> 16:17:34 INFO  main.c:56: Gain: 1.4 dB, Power: -41.733383 dBFS
> 16:17:34 INFO  main.c:56: Gain: 2.7 dB, Power: -38.573700 dBFS
> 16:17:34 INFO  main.c:56: Gain: 3.7 dB, Power: -36.965065 dBFS
> 16:17:35 INFO  main.c:56: Gain: 7.7 dB, Power: -33.935741 dBFS
> 16:17:35 INFO  main.c:56: Gain: 8.7 dB, Power: -32.194080 dBFS
> 16:17:35 INFO  main.c:56: Gain: 12.5 dB, Power: -29.703524 dBFS
> 16:17:35 INFO  main.c:56: Gain: 14.4 dB, Power: -28.519672 dBFS
> 16:17:35 INFO  main.c:56: Gain: 15.7 dB, Power: -26.887764 dBFS
> 16:17:35 INFO  main.c:56: Gain: 16.6 dB, Power: -25.499348 dBFS
> 16:17:35 INFO  main.c:56: Gain: 19.7 dB, Power: -22.934326 dBFS
> 16:17:35 INFO  main.c:56: Gain: 20.7 dB, Power: -21.509243 dBFS
> 16:17:35 INFO  main.c:56: Gain: 22.9 dB, Power: -19.128836 dBFS
> 16:17:35 INFO  main.c:56: Gain: 25.4 dB, Power: -17.998165 dBFS
> 16:17:35 INFO  main.c:56: Gain: 28.0 dB, Power: -16.607821 dBFS
> 16:17:35 INFO  main.c:56: Gain: 29.7 dB, Power: -15.704910 dBFS
> 16:17:35 INFO  main.c:56: Gain: 32.8 dB, Power: -14.389134 dBFS
> 16:17:35 INFO  main.c:56: Gain: 33.8 dB, Power: -13.820263 dBFS
> 16:17:35 INFO  main.c:56: Gain: 36.4 dB, Power: -13.252552 dBFS
> 16:17:35 INFO  main.c:56: Gain: 37.2 dB, Power: -12.886604 dBFS
> 16:17:35 INFO  main.c:56: Gain: 38.6 dB, Power: -12.515770 dBFS
> 16:17:35 INFO  main.c:56: Gain: 40.2 dB, Power: -12.257940 dBFS
> 16:17:35 INFO  main.c:56: Gain: 42.1 dB, Power: -12.139076 dBFS
> 16:17:36 INFO  main.c:56: Gain: 43.4 dB, Power: -12.042390 dBFS
> 16:17:36 INFO  main.c:56: Gain: 43.9 dB, Power: -11.806576 dBFS
> 16:17:36 INFO  main.c:56: Gain: 44.5 dB, Power: -11.831352 dBFS
> 16:17:36 INFO  main.c:56: Gain: 48.0 dB, Power: -11.845275 dBFS
> 16:17:36 INFO  main.c:56: Gain: 49.6 dB, Power: -11.956553 dBFS
> 16:17:36 DEBUG main.c:75: Best gain: 496

Here is a FFT at 20.7 dB which is basically what i posted earlier. 

[image]

And here is max gain 49.6 dB. It's very badly overloaded and possibly can't be a good gain level.

[image]

argilo, 2017-09-30:
Thanks, that definitely helps. Can you also check at what gain level the overloading starts? I.e. when does the noise floor start to rise up and swallow the HD sidebands?

pclov3r, 2017-09-30:
At 28dB the upper sideband is completely gone.

pclov3r, 2017-09-30:
Forgot to include I am using a dipole antenna that is built for the FM broadcast band and a RTL-SDR Blog v3 dongle.

pclov3r, 2017-09-30:
Just found another station that doesn't work actually, 

FFT at 20.7 dB 
[image]

argilo, 2017-09-30:
Thanks.

I think this is happening because the new AGC algorithm just tries to get the signal up to -10 dBFS at the ADC input. My assumption was that as long as the ADC was not overloaded, all would be well. But it seems you're getting overload somewhere earlier, and the signal to the ADC never reaches -10 dBFS so the AGC code just keeps cranking up the gain trying to reach that point.

I think I might put back the old FFT-based algorithm that measures SNR, but keep it at the current faster speed.

pclov3r, 2017-09-30:
That makes sense.

I guess it's going to dependent on which rtl-sdr dongle in use and your antenna. 

It's quite easy to overload the rtl-sdr on the FM broadband band with an efficient antenna in my case a dipole built for FM Broadcast so that may be the cause of the problems.

argilo, 2017-10-01:
Fixed by #82.

closed 2017-10-01 by argilo