theori-io/nrsc5 #86: rtl_tcp support

pclov3r, 2017-10-04:
So I thought it might be useful to have rtl_tcp support built in.

argilo, 2017-10-26:
By the way, you can do this today by running:
```
rtl_tcp -a 0.0.0.0 -f 100.3M -s 1488375 -g 20
```
```
nc 192.168.1.111 1234 | src/nrsc5 -r - 0
```
Building support into nrsc5 would save you from having to specify the frequency, sample rate and gain on the server, and from having to use `nc` on the client.

pclov3r, 2017-10-26:
@argilo Interesting. Looks useful for now.

People that have a rtl-sdr in some remote location may benefit form having this built in.

pclov3r, 2017-10-27:
@argilo your suggestion works for me and it even works on Windows as well.

adrianboyko, 2019-10-02:
Voting for this enhancement. I use multiple clients that share an rtl_tcp running on a Raspberry Pi. Needing to rerun rtl_tcp on the Pi, every time I want to listen to nrsc5 on my PC, is inconvenient.

fanlessfan, 2020-04-27:
Voting for this enhancement.

awesie, 2020-04-27:
Is the specific request to have nrsc5 connect to a rtl_tcp server, modify the server's current frequency and gain, and decode the audio?

fanlessfan, 2020-04-28:
That’s more convenient than login to the server to run the rtf-tcp.

PR #228 (2020-06-26): Basic rtl_tcp support
Fixes #86  

I quickly threw this together so there might be some tweaking we want to do on it. I tested the basic functionality with a Linux rtl_tcp server and nrsc5 clients on each OS.

Future improvements:

- optionally set ppm error (since it is likely already set on the server-side)
- auto-gain support (We know the number of gain levels, and we can iterate over them by index. We do not know the actual gain themselves though.)
closed 2020-06-26 by awesie