theori-io/nrsc5 #87: How to tell if there's an HD signal?

classicfm, 2017-10-10:
I love that someone has finally done this, so first of all, congratulations on this fine work. I use RTL SDR with analog FM already on a daily basis so I'm really hoping to just drop in your program to replace the FM SDR program I use already. However, I am having a problem. The station I care about most is our local classical station. They are HD-enabled, but I am not getting the signal. The analog signal comes in fine. I have _no_ other HD Radio to verify things with: one car we have is a 2009 and the other is a 2003(!!). I thought I remembered hearing they had a technical failure that affected HD but this has been months ago. I look at an FFT graph and see no "lobes". On other stations, I see them. Is there any way to get an idea of if a station has HD, and the problem is with me and my reception, or if there is no signal right now and I just have to wait until they come back on air?

fqrley, 2017-10-10:
It's a bit risky, but you might try contacting the radio station. 

Matt

pclov3r, 2017-10-10:
Not sure why it's risky to ask a station if they have HD radio service or not.

But on a FFT graph you should see two COFDM sidebands and you don't see that it means that they don't have HD radio service or it's not enabled. It could also mean weak or poor reception but this case the analog FM signal would be weak as well as well most likely.

classicfm, 2017-10-10:
He's being snarky, of course I could contact the station. 

The signal isn't the strongest but it's not particularly weak. According to FM Fool, it's about the same strength as our NPR affiliate, which works very well with this program.

I was mostly wondering if the CNR measurements might indicate if there was anything present in the sidebands, or if those numbers are measuring the FM channel as a whole. The FFT shows _nothing_. Not weak, but nothing at all. 

My understanding is that there's a maximum ERP for HD that is a percentage of the analog signal. It used to be lower than it is. I think a lot of stations have increased but maybe these folks haven't.

argilo, 2017-10-10:
The CNR measurement reported during automatic gain selection is for the HD Radio sidebands only. The power of the sidebands is compared to the power present at frequencies just above the upper HD sideband and just below the lower HD sideband.

classicfm, 2017-10-11:
Great, thanks. I think that answers my question. CNR numbers are rarely higher than 1 dB, so I will try moving my antenna around but it sounds to me like there's no HD transmission atm.

Thanks again for this great effort. I've been waiting for something like this to turn up for a while now.

closed 2017-10-11 by classicfm