theori-io/nrsc5 #88: Sound artifacts

classicfm, 2017-10-11:
I have come across an HD2 stream which exhibits some strange decoding artifacts. The stream seems to be lower bit rate than most other HD2 streams (it is 20-24 Kbps). I have attached the HDC stream, so it is hopefully easier to see what is different in this stream than using a WAV or ADTS output. I have a copy of the samples (-w option) but for 30 seconds the file is 102 MB even with XZ compression. Here is the log from nrsc5:

andrew@cheetah:~/Source/nrsc5$ ./src/nrsc5 -w waer2.bin -o waer2.hdc -f hdc 88.3 1
15:49:20 INFO  main.c:195: [0] Generic RTL2832U OEM
Found Rafael Micro R820T tuner 
Exact sample rate is: 1488375.071248 Hz
[R82XX] PLL not locked!
15:49:21 INFO  main.c:65: Gain: 0.0 dB, CNR: 3.881641 dB
15:49:21 INFO  main.c:65: Gain: 0.9 dB, CNR: 4.964504 dB
15:49:21 INFO  main.c:65: Gain: 1.4 dB, CNR: 5.897870 dB
15:49:21 INFO  main.c:65: Gain: 2.7 dB, CNR: 7.068104 dB
15:49:21 INFO  main.c:65: Gain: 3.7 dB, CNR: 8.563585 dB
15:49:21 INFO  main.c:65: Gain: 7.7 dB, CNR: 10.021370 dB
15:49:21 INFO  main.c:65: Gain: 8.7 dB, CNR: 11.230439 dB
15:49:21 INFO  main.c:65: Gain: 12.5 dB, CNR: 12.968244 dB
15:49:21 INFO  main.c:65: Gain: 14.4 dB, CNR: 13.270781 dB
15:49:21 INFO  main.c:65: Gain: 15.7 dB, CNR: 13.878690 dB
15:49:21 INFO  main.c:65: Gain: 16.6 dB, CNR: 14.144740 dB
15:49:21 INFO  main.c:65: Gain: 19.7 dB, CNR: 15.064043 dB
15:49:21 INFO  main.c:65: Gain: 20.7 dB, CNR: 15.176182 dB
15:49:21 INFO  main.c:65: Gain: 22.9 dB, CNR: 15.483320 dB
15:49:21 INFO  main.c:65: Gain: 25.4 dB, CNR: 15.320906 dB
15:49:21 INFO  main.c:65: Gain: 28.0 dB, CNR: 13.060275 dB
15:49:21 INFO  main.c:65: Gain: 29.7 dB, CNR: 10.318936 dB
15:49:21 DEBUG main.c:69: Best gain: 229
15:49:24 INFO  acquire.c:104: Timing offset: 285.687500, slope: 4.707031 (adjust)
15:49:25 INFO  input.c:163: CFO: 6540.710449 Hz (74 ppm)
15:49:25 DEBUG sync.c:270: First block @ 27
15:49:26 DEBUG sync.c:270: First block @ 0
15:49:26 INFO  sync.c:248: Synchronized!
15:49:26 DEBUG pids.c:136: Station Name: WAER
15:49:26 DEBUG pids.c:119: Country: US, FCC facility ID: 64354
15:49:26 DEBUG pids.c:270: Audio program 0: public, type 0, sound experience 0
15:49:26 DEBUG pids.c:270: Audio program 1: public, type 0, sound experience 0
15:49:27 DEBUG pids.c:270: Audio program 2: public, type 0, sound experience 0
15:49:27 DEBUG pids.c:196: Station location: 43.033691 -179.869019, 192m
15:49:27 DEBUG pids.c:398: Slogan: WAER
15:49:27 DEBUG pids.c:169: Long station name: WAER
15:49:28 INFO  sync.c:324: MER: 11.789657 dB (lower), 12.490125 dB (upper)
15:49:28 INFO  decode.c:77: BER: 0.000306, avg: 0.000306, min: 0.000306, max: 0.000306
15:49:28 DEBUG output.c:382: Title: Jazz on WAER
15:49:28 DEBUG frame.c:465: ignoring partial pdu
15:49:29 DEBUG pids.c:317: Pending leap second offset: 0, current leap second offset: 18
15:49:29 INFO  sync.c:324: MER: 12.724352 dB (lower), 13.212567 dB (upper)
15:49:29 INFO  decode.c:77: BER: 0.000249, avg: 0.000278, min: 0.000249, max: 0.000306
15:49:29 DEBUG output.c:382: Title: Jazz on WAER
15:49:29 DEBUG output.c:127: Audio bit rate: 21.3 kbps
15:49:31 DEBUG acquire.c:99: Timing offset: 383.312500, slope: 0.355469
15:49:31 DEBUG pids.c:322: ALFN of pending leap second adjustment: 0
15:49:31 DEBUG pids.c:328: Local time zone offset: -480 minutes, DST sched. 1, local DST? yes, regional DST? yes
15:49:31 INFO  sync.c:324: MER: 12.539015 dB (lower), 13.263510 dB (upper)
15:49:31 INFO  decode.c:77: BER: 0.000224, avg: 0.000260, min: 0.000224, max: 0.000306
15:49:31 DEBUG output.c:382: Title: Jazz on WAER
15:49:31 DEBUG output.c:127: Audio bit rate: 22.1 kbps
15:49:32 INFO  sync.c:324: MER: 12.628249 dB (lower), 13.230733 dB (upper)
15:49:32 INFO  decode.c:77: BER: 0.000252, avg: 0.000258, min: 0.000224, max: 0.000306
15:49:32 DEBUG output.c:382: Title: Jazz on WAER
15:49:32 DEBUG output.c:382: Title: Jazz on WAER
15:49:32 DEBUG output.c:127: Audio bit rate: 23.1 kbps
15:49:34 DEBUG acquire.c:99: Timing offset: 394.687500, slope: 0.332031
15:49:34 INFO  sync.c:324: MER: 12.784842 dB (lower), 13.333170 dB (upper)
15:49:34 INFO  decode.c:77: BER: 0.000241, avg: 0.000254, min: 0.000224, max: 0.000306
15:49:34 DEBUG output.c:382: Title: Jazz on WAER
15:49:34 DEBUG output.c:127: Audio bit rate: 21.9 kbps
15:49:35 INFO  sync.c:324: MER: 12.858532 dB (lower), 13.515463 dB (upper)
15:49:35 INFO  decode.c:77: BER: 0.000263, avg: 0.000256, min: 0.000224, max: 0.000306
15:49:35 INFO  frame.c:331: Subchannel 0: mode=0, length=937
15:49:35 INFO  frame.c:331: Subchannel 1: mode=0, length=0
15:49:35 INFO  frame.c:331: Subchannel 2: mode=0, length=0
15:49:35 DEBUG output.c:499: missing port 0402
[... 77 lines cut ...]
15:49:56 DEBUG output.c:382: Title: Jazz on WAER from Syracuse University
15:49:56 DEBUG output.c:127: Audio bit rate: 22.2 kbps
15:49:57 DEBUG acquire.c:99: Timing offset: 482.437500, slope: 0.285156
15:49:58 INFO  sync.c:324: MER: 12.401589 dB (lower), 12.894782 dB (upper)
15:49:58 INFO  decode.c:77: BER: 0.000249, avg: 0.000242, min: 0.000170, max: 0.000306
15:49:58 DEBUG output.c:382: Title: Jazz on WAER from Syracuse University
15:49:58 DEBUG output.c:127: Audio bit rate: 22.5 kbps
15:49:59 INFO  sync.c:324: MER: 12.709877 dB (lower), 13.236389 dB (upper)
15:49:59 INFO  decode.c:77: BER: 0.000219, avg: 0.000241, min: 0.000170, max: 0.000306
15:49:59 DEBUG output.c:382: Title: Jazz on WAER from Syracuse University
15:49:59 DEBUG output.c:127: Audio bit rate: 22.1 kbps
15:50:00 DEBUG acquire.c:99: Timing offset: 493.000000, slope: 0.312500
15:50:00 INFO  sync.c:324: MER: 12.887310 dB (lower), 13.420171 dB (upper)
15:50:00 INFO  decode.c:77: BER: 0.000282, avg: 0.000243, min: 0.000170, max: 0.000306
15:50:00 DEBUG output.c:382: Title: Jazz on WAER from Syracuse University
15:50:00 DEBUG output.c:127: Audio bit rate: 21.7 kbps
15:50:02 INFO  sync.c:324: MER: 12.746115 dB (lower), 13.331119 dB (upper)
15:50:02 INFO  decode.c:77: BER: 0.000189, avg: 0.000240, min: 0.000170, max: 0.000306
15:50:02 DEBUG output.c:522: 00000001 7EBA5CF1 1E653E9C
15:50:02 INFO  output.c:546: File SLWAER  010100HD1.jpg, size 11072
15:50:02 DEBUG output.c:382: Title: Jazz on WAER from Syracuse University
15:50:02 DEBUG output.c:127: Audio bit rate: 22.4 kbps
[Uploading waer2.zip…]()

[Uploading waer2.zip…]()

pclov3r, 2017-10-12:
I think the issue is just the low bit-rate. 24 Kbps with music isn't going to sound very good.

classicfm, 2017-10-12:
I agree that the bit rate is too low (HD Radio is kind of a disappointment so far, IMO, at least compared to FM Stereo. Kind of sad that after almost 60 years we couldn't have done better.) But did you listen to the sample I posted? I have a hard time believing that's just the failure mode of the codec. It certainly could be, but to my untrained ear it sounds like there are frames of audio that are being played at the wrong point, interleaved with frames that belong at that point in time. There are also some points at which it makes a rather unpleasant, but short-lived, high pitched noise, like some corrupted MP3s used to do years ago. I'm not sure that that's in the sample I posted, but it did occur during the testing I was doing yesterday.

It could be the radio station's fault, but I have a somewhat hard time believing it's "supposed" to be like that. But I could be wrong!

pclov3r, 2017-10-13:
Can you a post a sample created with NRSC5 using the -w flag and not an audio output?

argilo, 2017-10-13:
@classicfm I converted your HDC file to ADTS using hdc_to_aac.c, and noticed that it sounded very good when played in Audacity but awful when played in VLC. If nrsc5 is using the same decoder as VLC, that could explain the poor audio quality. It could be that switching from libfaad2 to fdk-aac would solve the problem.

It would be helpful if you could make a recording of the RF signal using `-w`, like @pclov3r suggested.

pclov3r, 2017-10-13:
@argilo I'm curious how you converted that file? Or if you could post the converted ADTS stream file that would be great as I'm curious myself.

argilo, 2017-10-13:
@pclov3r I used the following to check out an old version of hdc_to_aac.c that still had a `main` function:

`git checkout a4e4327b577cf0282fd3d3a5eaec71d9ad6e9087 src/hdc_to_aac.c`

Then I made a couple tweaks to the file:

```diff
diff --git a/src/hdc_to_aac.c b/src/hdc_to_aac.c
index 6f4737a..1ce948e 100644
--- a/src/hdc_to_aac.c
+++ b/src/hdc_to_aac.c
@@ -1099,11 +1099,10 @@ void hdc_to_aac(bitreader_t *br, bitwriter_t *bw)
     parse_packet(br, bw);
 }
 
-#if 0
 int main(int argc, char *argv[])
 {
     FILE *fp = fopen(argv[1], "rb");
-    FILE *fout = fopen("test.adts", "wb");
+    FILE *fout = fopen(argv[2], "wb");
 
     while (1)
     {
@@ -1132,4 +1131,3 @@ int main(int argc, char *argv[])
 
     return 0;
 }
-#endif
```

Compiled it:

`gcc -o hdc_to_aac src/hdc_to_aac.c`

and ran:

`./hdc_to_aac waer2.hdc waer.adts`

pclov3r, 2017-10-13:
@argilo thanks!

VLC on windows sounds awful as well but for some reason it sounds fine with Winamp.

argilo, 2017-10-13:
It appears that VLC uses FAAD2, just like nrsc5: https://wiki.videolan.org/Advanced+Audio+Coding

My guess would be that it doesn't support all the codec features used by HDC.

argilo, 2017-10-13:
Audacity seems to use ffmpeg, which decodes HE-AAC using fdk-aac. So I suspect that switching nrsc5 to fdk-aac would solve the problem. Its decoder would need to be modified to handle HDC; I've only modified the encoder so far: https://github.com/mstorsjo/fdk-aac/compare/master...argilo:hdc-encoder

I suspect that switching to fdk-aac would also solve #53.

classicfm, 2017-10-13:
I have a recording which was taken simultaneously with the sound sample I posted, this is the waer2.bin referred to in the command line of my original report. It is 102 MB with XZ compression as I noted: http://www-ext.ajb.ddns.us/public/waer2.bin.xz

I should point out that fdk-aac is considered non-free by Debian, presumably due to the last sentence in section 3, which might be problematic for this project as well: http://metadata.ftp-master.debian.org/changelogs/non-free/f/fdk-aac/fdk-aac_0.1.4-2_copyright

awesie, 2017-10-14:
I think the main difference between ffmpeg and us is that it properly ignores SBR extensions that it cannot understand.

I fixed that bug in the patch to FAAD2. Please pull the latest experimental branch, and recompile. You may need to delete your existing faad2-prefix directory.

classicfm, 2017-10-14:
The artifacts are gone, but the FFmpeg decoder still produces higher quality output. Now, FAAD2 sounds like you might expect a 24 kbps stream to sound (muddy on the high end), but FFmpeg, presumably understanding more SBR objects than FAAD2, is able to output a .wav file with a much nicer sounding high range.

But in any case, the artifacts and glitching is gone.

argilo, 2017-10-14:
Sounds much better now, @awesie. :+1:

awesie, 2017-10-14:
It seems like HDC uses similar PS as DRM. So right now the PS data is ignored resulting in a mono sound (e.g. you lose a channel).

I hacked together the DRM PS support with HDC. I think it sounds a little bit better (https://drive.google.com/file/d/0B6kMFgnRJ5AycWI3UHNaUG5rZ2M/view?usp=sharing). Thoughts?

pclov3r, 2017-10-14:
@awesie Does that mean currently all streams are mono only audio?

awesie, 2017-10-14:
@pclov3r I don't think so. Only the odd ones (e.g. low bitrate music) that make use of the PS extension.

pclov3r, 2017-10-14:
@awesie Hum, Any clues as to what the output message would be if the PS extension is in use? 

Curious now.

classicfm, 2017-10-14:
That .wav file sounds about right. I don't have fdk-aac so I will leave it to others to judge if FAAD2+PS is as good as the output of the Fraunhofer codec.

argilo, 2017-10-15:
@awesie The left channel in your WAV file sounds good, but the right channel has a clicking / pulsing sound, perhaps at 44100 / 2048 = 21.5 Hz?

classicfm, 2017-10-16:
I'm satisfied with the new patch. Thank you so much! HD Radio is even more like DRM than I thought. Which makes its proprietary nature even more frustrating. But I digress. If you folks are happy, go ahead and close this bug.

Thanks again for the great work! Hopefully our classical radio station gets HD back soon. Lightning strike at the antenna site last month.

classicjazz, 2017-10-18:
Not sure if this is related to this patch but thought I would post what I am seeing piping ADTS to MPV using experimental branch. I built in the last hour:

Tuning to the radio station: 90.9 Mhz HD1
   cplayer: Playing: -
      file: Reading from stdin...
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
     demux: Not seekable, but enabling seeking on user request.
   cplayer:  (+) Audio --aid=1 (aac 2ch 44100Hz)
   cplayer: AO: [coreaudio] 48000Hz stereo 2ch float
statusline: A: 00:00:47 / 00:00:00 Cache:  1s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:00:52 / 00:00:00 Cache:  5s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 254 is invalid
statusline: A: 00:00:54 / 00:00:00 Cache:  5s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: Number of bands (57) exceeds limit (45).
        ad: Error decoding audio.
statusline: A: 00:00:54 / 00:00:00 Cache:  4s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:00:55 / 00:00:00 Cache:  3s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 252 is invalid
statusline: A: 00:00:58 / 00:00:00 Cache:  0s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:01:01 / 00:00:00 Cache:  9s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:01:05 / 00:00:00 Cache:  7s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: invalid band type
        ad: Error decoding audio.
statusline: A: 00:01:10 / 00:00:00 Cache:  7s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: Reserved SBR extensions is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.
ffmpeg/audio: aac: If you want to help, upload a sample of this file to ftp://upload.ffmpeg.org/incoming/ and contact the ffmpeg-devel mailing list. (ffmpeg-devel@ffmpeg.org)
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
    ffmpeg: envelope scalefactor overflow in dequant
statusline: A: 00:01:20 / 00:00:00 Cache: 10s+10KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:01:22 / 00:00:00 Cache: 10s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: Number of bands (56) exceeds limit (45).
        ad: Error decoding audio.
statusline: A: 00:01:22 / 00:00:00 Cache: 10s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:01:23 / 00:00:00 Cache:  9s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:01:25 / 00:00:00 Cache: 10s+0KB
[+-----------------------------------------------------------------------------]
ffmpeg/audio: aac: env_facs_q 255 is invalid
statusline: A: 00:01:27 / 00:00:00 Cache:  9s+0KB
^C-----------------------------------------------------------------------------]

awesie, 2017-10-18:
I think that is the same as #53, which is unrelated to this issue and related patches. The commits for this issue only affect output through the built-in FAAD2 library; they do not affect ADTS output.

If you are listening to stream that uses the DRM PS extension with ADTS output, you may get the following output repeatedly:

> [ffmpeg/audio] aac: Reserved SBR extensions is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.

As the error message explains, this is due to most versions of FFmpeg not supporting DRM PS with AAC-SBR.

Any other errors, e.g. env_facs_q 255 is invalid, are likely due to corruption of the signal. If you post a samples file we can test in #53, it would be helpful.

classicjazz, 2017-10-19:
Samples posted in #53. FFMPEG 3.4 details shared there. 

My takeaway from this issue is that users should use WAV output to ensure SBR decoding and therefore optimal sound fidelity. Correct?

awesie, 2019-02-12:
I think this issue is deprecated now that the ADTS output is deprecated. If there are any more sound quality issues with the built-in decoder, they should get a new issue.

closed 2019-02-12 by awesie