theori-io/nrsc5 #92: Lots of CRC mismatches

impala454, 2017-10-21:
Hello, I'm a long time software developer but brand new to SDR and just learning.  I have been messing with nrsc5 for about a week now and keep having trouble getting it going.  I have a NooElec NESDR Mini 2 w/the default antenna and am building and running on Ubuntu 16.04.  I've chosen one of the stronger stations in the Houston area and keep getting CRC mismatches which end up exiting the program.  I'm running in a VMware VM but have also tried it on bare metal.  I've loaded a fresh Ubuntu 16.04 image with nothing but the dependencies and this software and still get the same issue.  I notice all of the extra data is found (even the png image!) but it seems when it gets to the audio decode something gets messed up.  As a newbie I'm not sure if my SNR is high enough but this channel was the highest I saw in my area, which I'd expect to be good as it's a huge city.  Any guidance would be appreciated or if others aren't using Ubuntu then maybe I found something.  Please let me know if there's something I can do to provide more data to debug with.  Here's a sample run:

```
22:33:10:chuck@chux-xenial-base:build:master✔  ►  src/nrsc5 -o - -f adts 99.1 0 | mplayer -
22:33:15 INFO  main.c:195: [0] Generic RTL2832U OEM
MPlayer 1.2.1 (Debian), built with gcc-5.3.1 (C) 2000-2016 MPlayer Team
mplayer: could not connect to socket
mplayer: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing -.
Reading from stdin...
libavformat version 56.40.101 (external)
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
[R82XX] PLL not locked!
22:33:16 INFO  main.c:65: Gain: 0.0 dB, CNR: 0.126053 dB
22:33:16 INFO  main.c:65: Gain: 0.9 dB, CNR: 0.102925 dB
22:33:16 INFO  main.c:65: Gain: 1.4 dB, CNR: 0.128405 dB
22:33:16 INFO  main.c:65: Gain: 2.7 dB, CNR: 0.640147 dB
22:33:16 INFO  main.c:65: Gain: 3.7 dB, CNR: 0.671004 dB
22:33:16 INFO  main.c:65: Gain: 7.7 dB, CNR: 1.523039 dB
22:33:17 INFO  main.c:65: Gain: 8.7 dB, CNR: 1.547986 dB
22:33:17 INFO  main.c:65: Gain: 12.5 dB, CNR: 2.083460 dB
22:33:17 INFO  main.c:65: Gain: 14.4 dB, CNR: 2.498631 dB
22:33:17 INFO  main.c:65: Gain: 15.7 dB, CNR: 2.579844 dB
22:33:17 INFO  main.c:65: Gain: 16.6 dB, CNR: 2.973796 dB
22:33:17 INFO  main.c:65: Gain: 19.7 dB, CNR: 4.781415 dB
22:33:17 INFO  main.c:65: Gain: 20.7 dB, CNR: 5.514703 dB
22:33:17 INFO  main.c:65: Gain: 22.9 dB, CNR: 6.850557 dB
22:33:17 INFO  main.c:65: Gain: 25.4 dB, CNR: 6.483753 dB
22:33:17 INFO  main.c:65: Gain: 28.0 dB, CNR: 7.485606 dB
22:33:17 INFO  main.c:65: Gain: 29.7 dB, CNR: 7.307911 dB
22:33:17 INFO  main.c:65: Gain: 32.8 dB, CNR: 8.017851 dB
22:33:17 INFO  main.c:65: Gain: 33.8 dB, CNR: 8.378475 dB
22:33:17 INFO  main.c:65: Gain: 36.4 dB, CNR: 7.912903 dB
22:33:17 INFO  main.c:65: Gain: 37.2 dB, CNR: 8.228477 dB
22:33:17 INFO  main.c:65: Gain: 38.6 dB, CNR: 8.047071 dB
22:33:18 INFO  main.c:65: Gain: 40.2 dB, CNR: 7.731998 dB
22:33:18 INFO  main.c:65: Gain: 42.1 dB, CNR: 8.289161 dB
22:33:18 INFO  main.c:65: Gain: 43.4 dB, CNR: 7.929200 dB
22:33:18 INFO  main.c:65: Gain: 43.9 dB, CNR: 7.954932 dB
22:33:18 INFO  main.c:65: Gain: 44.5 dB, CNR: 7.988502 dB
22:33:18 INFO  main.c:65: Gain: 48.0 dB, CNR: 8.573486 dB
22:33:18 INFO  main.c:65: Gain: 49.6 dB, CNR: 7.985606 dB
22:33:18 DEBUG main.c:69: Best gain: 480
22:33:21 INFO  acquire.c:104: Timing offset: 431.875000, slope: 3.507812 (adjust)
22:33:22 INFO  input.c:163: CFO: 5450.591797 Hz (55 ppm)
22:33:22 DEBUG sync.c:270: First block @ 17
22:33:23 DEBUG sync.c:270: First block @ 0
22:33:24 INFO  sync.c:248: Synchronized!
22:33:24 DEBUG pids.c:270: Audio program 2: public, type 18, sound experience 0
22:33:24 DEBUG pids.c:292: Data service: public, type 65, MIME type 5b6
22:33:24 DEBUG pids.c:136: Station Name: KODA-FM
22:33:24 DEBUG pids.c:119: Country: US, FCC facility ID: 35337
22:33:25 DEBUG pids.c:292: Data service: public, type 66, MIME type 2d7
22:33:25 DEBUG pids.c:292: Data service: public, type 66, MIME type e96
22:33:25 DEBUG pids.c:398: Slogan: SUNNY 99.1
22:33:25 DEBUG pids.c:238: Message (priority 0):  
22:33:25 INFO  sync.c:324: MER: -7.903810 dB (lower), -7.124453 dB (upper)
22:33:25 DEBUG pids.c:270: Audio program 0: public, type 8, sound experience 0
22:33:25 DEBUG pids.c:270: Audio program 1: public, type 3, sound experience 0
22:33:25 INFO  decode.c:77: BER: 0.155210, avg: 0.155210, min: 0.155210, max: 0.155210
22:33:26 DEBUG pids.c:292: Data service: public, type 259, MIME type eb2
22:33:26 DEBUG pids.c:292: Data service: public, type 259, MIME type 469
22:33:26 DEBUG pids.c:196: Station location: 99.214600 -244.031372, 128m
22:33:27 INFO  sync.c:324: MER: -3.523866 dB (lower), -0.957806 dB (upper)
22:33:27 INFO  decode.c:77: BER: 0.055798, avg: 0.105504, min: 0.055798, max: 0.155210
22:33:27 DEBUG acquire.c:99: Timing offset: 504.375000, slope: 0.164062
22:33:28 INFO  sync.c:324: MER: -7.338721 dB (lower), 6.083565 dB (upper)
[... 101 lines cut ...]
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed

Exiting... (End of file)```

pclov3r, 2017-10-21:
From looking at the output the CRC errors are caused by very low SNR.

The antenna that comes with that isn't very good so I would try to build a dipole for the FM broadcast band and try again. Rabbit ears are also a good antenna for FM broadcast band.

impala454, 2017-10-21:
Ok thanks.  About what SNR am I looking for?

pclov3r, 2017-10-21:
Are you able to post a FFT shot of this station form software such as GQRX?

impala454, 2017-10-21:
[image]

pclov3r, 2017-10-21:
@impala454 Looking at that screenshot the COFDM sidebands are basically burred in noise which explains your issues.

I would try a different antenna and try again.

impala454, 2017-10-21:
Ok will do.  I also went ahead and used SDR# to find a better station (where the sidebands were easily visible) then went back and tried nrsc5 again.  This time it appears to work, however I still get no audio.  Here's a snippet of that log:

```
23:09:36 DEBUG output.c:127: Audio bit rate: 63.5 kbps
23:09:37 DEBUG acquire.c:99: Timing offset: 766.750000, slope: 0.453125
23:09:38 INFO  sync.c:324: MER: 4.440616 dB (lower), 6.667902 dB (upper)
23:09:38 INFO  decode.c:77: BER: 0.007867, avg: 0.011872, min: 0.005147, max: 0.075052
23:09:38 INFO  output.c:561: Received 2a69a04b5082c6958423f021fe4e5dfe.jpg
23:09:38 DEBUG output.c:382: Title: I Don't Wanna Live Forever
23:09:38 DEBUG output.c:388: Artist: ZAYN & Taylor Swift
23:09:38 DEBUG output.c:394: Album: I Don't Wanna Live Forever [Single]
23:09:38 DEBUG output.c:127: Audio bit rate: 62.7 kbps
23:09:39 INFO  sync.c:324: MER: 6.377982 dB (lower), 6.779168 dB (upper)
23:09:39 INFO  decode.c:77: BER: 0.006234, avg: 0.011773, min: 0.005147, max: 0.075052
23:09:39 DEBUG output.c:127: Audio bit rate: 63.4 kbps
23:09:40 DEBUG acquire.c:99: Timing offset: 779.562500, slope: 0.402344
23:09:41 INFO  sync.c:324: MER: 3.877916 dB (lower), 6.979779 dB (upper)
23:09:41 INFO  decode.c:77: BER: 0.006168, avg: 0.011677, min: 0.005147, max: 0.075052
23:09:41 DEBUG output.c:382: Title: I Don't Wanna Live Forever
23:09:41 DEBUG output.c:388: Artist: ZAYN & Taylor Swift
23:09:41 DEBUG output.c:394: Album: I Don't Wanna Live Forever [Single]
23:09:41 DEBUG output.c:127: Audio bit rate: 63.4 kbps
23:09:42 INFO  sync.c:324: MER: 3.824472 dB (lower), 6.963761 dB (upper)
23:09:42 INFO  decode.c:77: BER: 0.005473, avg: 0.011571, min: 0.005147, max: 0.075052
23:09:42 DEBUG output.c:127: Audio bit rate: 62.0 kbps
23:09:43 DEBUG acquire.c:99: Timing offset: 790.812500, slope: 0.636719
23:09:44 INFO  sync.c:324: MER: 5.692590 dB (lower), 6.043542 dB (upper)
23:09:44 INFO  decode.c:77: BER: 0.008450, avg: 0.011519, min: 0.005147, max: 0.075052
23:09:44 DEBUG output.c:522: 00000001 7EBAA909 1E653E9C
23:09:44 INFO  output.c:546: File 2a69a04b5082c6958423f021fe4e5dfe.jpg, size 6894
23:09:44 DEBUG output.c:382: Title: I Don't Wanna Live Forever
23:09:44 DEBUG output.c:388: Artist: ZAYN & Taylor Swift
23:09:44 DEBUG output.c:394: Album: I Don't Wanna Live Forever [Single]
23:09:44 DEBUG output.c:127: Audio bit rate: 65.1 kbps
23:09:45 INFO  sync.c:324: MER: 1.830562 dB (lower), 6.595309 dB (upper)
23:09:45 INFO  decode.c:77: BER: 0.007791, avg: 0.011458, min: 0.005147, max: 0.075052
23:09:45 DEBUG output.c:127: Audio bit rate: 61.2 kbps
23:09:47 INFO  sync.c:324: MER: 6.314525 dB (lower), 7.045503 dB (upper)
23:09:47 INFO  decode.c:77: BER: 0.005309, avg: 0.011359, min: 0.005147, max: 0.075052
23:09:47 DEBUG output.c:382: Title: I Don't Wanna Live Forever
23:09:47 DEBUG output.c:388: Artist: ZAYN & Taylor Swift
23:09:47 DEBUG output.c:394: Album: I Don't Wanna Live Forever [Single]
23:09:47 DEBUG output.c:127: Audio bit rate: 64.1 kbps
23:09:48 INFO  sync.c:324: MER: 6.363857 dB (lower), 7.116056 dB (upper)
23:09:48 INFO  decode.c:77: BER: 0.005092, avg: 0.011260, min: 0.005092, max: 0.075052
23:09:48 DEBUG output.c:127: Audio bit rate: 61.9 kbps
23:09:50 INFO  sync.c:324: MER: 6.776006 dB (lower), 6.990862 dB (upper)
23:09:50 INFO  decode.c:77: BER: 0.006009, avg: 0.011178, min: 0.005092, max: 0.075052
23:09:50 DEBUG output.c:382: Title: I Don't Wanna Live Forever
23:09:50 DEBUG output.c:388: Artist: ZAYN & Taylor Swift
23:09:50 DEBUG output.c:394: Album: I Don't Wanna Live Forever [Single]
23:09:50 DEBUG output.c:127: Audio bit rate: 64.0 kbps
```

pclov3r, 2017-10-21:
@impala454 Hum, How about running it directly and not piping it to mplayer?

impala454, 2017-10-21:
One more sample of another from the execution until I stopped it, and a screenshot of the FFT:

[image]

```
23:12:49:chuck@chux-xenial-base:build:master✔  ►  src/nrsc5 -o - -f adts 104.1 0 | mplayer -
23:13:56 INFO  main.c:195: [0] Generic RTL2832U OEM
MPlayer 1.2.1 (Debian), built with gcc-5.3.1 (C) 2000-2016 MPlayer Team

Playing -.
Reading from stdin...
libavformat version 56.40.101 (external)
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
[R82XX] PLL not locked!
23:13:58 INFO  main.c:65: Gain: 0.0 dB, CNR: -0.005706 dB
23:13:58 INFO  main.c:65: Gain: 0.9 dB, CNR: 0.465209 dB
23:13:58 INFO  main.c:65: Gain: 1.4 dB, CNR: 0.569711 dB
23:13:58 INFO  main.c:65: Gain: 2.7 dB, CNR: 0.850312 dB
23:13:58 INFO  main.c:65: Gain: 3.7 dB, CNR: 1.317207 dB
23:13:58 INFO  main.c:65: Gain: 7.7 dB, CNR: 2.191893 dB
23:13:58 INFO  main.c:65: Gain: 8.7 dB, CNR: 3.094390 dB
23:13:58 INFO  main.c:65: Gain: 12.5 dB, CNR: 3.656334 dB
23:13:58 INFO  main.c:65: Gain: 14.4 dB, CNR: 4.588475 dB
23:13:58 INFO  main.c:65: Gain: 15.7 dB, CNR: 4.994563 dB
23:13:58 INFO  main.c:65: Gain: 16.6 dB, CNR: 5.403742 dB
23:13:58 INFO  main.c:65: Gain: 19.7 dB, CNR: 7.164834 dB
23:13:58 INFO  main.c:65: Gain: 20.7 dB, CNR: 7.381543 dB
23:13:58 INFO  main.c:65: Gain: 22.9 dB, CNR: 8.528152 dB
23:13:59 INFO  main.c:65: Gain: 25.4 dB, CNR: 8.473986 dB
23:13:59 INFO  main.c:65: Gain: 28.0 dB, CNR: 9.362489 dB
23:13:59 INFO  main.c:65: Gain: 29.7 dB, CNR: 8.884693 dB
23:13:59 INFO  main.c:65: Gain: 32.8 dB, CNR: 9.305030 dB
23:13:59 INFO  main.c:65: Gain: 33.8 dB, CNR: 9.139647 dB
23:13:59 INFO  main.c:65: Gain: 36.4 dB, CNR: 9.340781 dB
23:13:59 INFO  main.c:65: Gain: 37.2 dB, CNR: 9.048660 dB
23:13:59 INFO  main.c:65: Gain: 38.6 dB, CNR: 9.546430 dB
23:13:59 INFO  main.c:65: Gain: 40.2 dB, CNR: 9.442328 dB
23:13:59 INFO  main.c:65: Gain: 42.1 dB, CNR: 9.238004 dB
23:13:59 INFO  main.c:65: Gain: 43.4 dB, CNR: 9.530404 dB
23:13:59 INFO  main.c:65: Gain: 43.9 dB, CNR: 8.896086 dB
23:13:59 INFO  main.c:65: Gain: 44.5 dB, CNR: 9.775655 dB
23:13:59 INFO  main.c:65: Gain: 48.0 dB, CNR: 9.375166 dB
23:13:59 INFO  main.c:65: Gain: 49.6 dB, CNR: 9.494567 dB
23:13:59 DEBUG main.c:69: Best gain: 445
23:14:03 INFO  acquire.c:104: Timing offset: 931.812500, slope: 3.761719 (adjust)
23:14:03 INFO  input.c:163: CFO: 5813.964844 Hz (56 ppm)
23:14:03 DEBUG sync.c:270: First block @ 21
23:14:04 DEBUG sync.c:270: First block @ 0
23:14:04 INFO  sync.c:248: Synchronized!
23:14:04 DEBUG pids.c:136: Station Name: KRBE-FM
23:14:04 DEBUG pids.c:119: Country: US, FCC facility ID: 35524
23:14:04 DEBUG pids.c:270: Audio program 0: public, type 0, sound experience 0
23:14:04 DEBUG pids.c:270: Audio program 1: public, type 0, sound experience 0
23:14:05 DEBUG pids.c:196: Station location: 29.576050 -160.489380, 0m
23:14:05 DEBUG pids.c:169: Long station name: MPS
23:14:05 DEBUG pids.c:238: Message (priority 0): 104.1 KRBE
23:14:05 INFO  sync.c:324: MER: 8.179256 dB (lower), 3.001954 dB (upper)
23:14:06 INFO  decode.c:77: BER: 0.016279, avg: 0.016279, min: 0.016279, max: 0.016279
23:14:06 DEBUG frame.c:465: ignoring partial pdu
23:14:06 DEBUG pids.c:398: Slogan: MPS
23:14:07 INFO  sync.c:324: MER: 8.217875 dB (lower), 2.193887 dB (upper)
23:14:07 INFO  decode.c:77: BER: 0.017322, avg: 0.016800, min: 0.016279, max: 0.017322
23:14:07 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:07 DEBUG output.c:388: Artist:  
23:14:07 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:07 DEBUG output.c:388: Artist:  
Cannot seek backward in linear streams!
Seek failed
23:14:08 INFO  sync.c:324: MER: 8.442745 dB (lower), -0.082278 dB (upper)
[... 67 lines cut ...]
23:14:20 DEBUG output.c:127: Audio bit rate: 46.6 kbps
23:14:22 INFO  sync.c:324: MER: 8.176266 dB (lower), 1.645356 dB (upper)
23:14:22 INFO  decode.c:77: BER: 0.016014, avg: 0.017198, min: 0.012744, max: 0.019056
23:14:22 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:22 DEBUG output.c:388: Artist:  
23:14:22 DEBUG output.c:127: Audio bit rate: 47.4 kbps
23:14:23 INFO  sync.c:324: MER: 8.250845 dB (lower), 2.691557 dB (upper)
23:14:23 INFO  decode.c:77: BER: 0.016807, avg: 0.017168, min: 0.012744, max: 0.019056
23:14:23 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:23 DEBUG output.c:388: Artist:  
23:14:23 DEBUG output.c:127: Audio bit rate: 47.5 kbps
23:14:25 INFO  sync.c:324: MER: 8.376478 dB (lower), 4.625756 dB (upper)
23:14:25 INFO  decode.c:77: BER: 0.012653, avg: 0.016846, min: 0.012653, max: 0.019056
23:14:25 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:25 DEBUG output.c:388: Artist:  
23:14:25 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:25 DEBUG output.c:388: Artist:  
23:14:25 DEBUG output.c:127: Audio bit rate: 47.1 kbps
23:14:26 INFO  sync.c:324: MER: 8.190929 dB (lower), 1.839442 dB (upper)
23:14:26 INFO  decode.c:77: BER: 0.017032, avg: 0.016858, min: 0.012653, max: 0.019056
23:14:26 DEBUG output.c:382: Title: Houston's #1 Hit Music Station
23:14:26 DEBUG output.c:388: Artist:  
23:14:26 DEBUG output.c:127: Audio bit rate: 46.4 kbps
^C
```

impala454, 2017-10-21:
Ran it without the pipe and see all kinds of binary data, so I suspect it's working?  Guess I'm not sure why the mplayer part isn't working.  I know when I run the included sample I hear audio.

pclov3r, 2017-10-21:
I wonder if it's possible the station in question has a broken dead-air stream if the included sample works.

Is it possible for you use post a sample from the station using the -w samples-output?

impala454, 2017-10-21:
Whoa... uh that makes me hear audio... well ok then.  I expected that command would just save samples.  I did this and now I hear it loud and clear through the speakers:
`src/nrsc5 -w hou1041fm.samples -f adts 104.1 0`

impala454, 2017-10-21:
Was my syntax above wrong?

pclov3r, 2017-10-21:
Try running the command without piping to mplayer

try using the command `nrsc5 104.1 0` and see what happens.

impala454, 2017-10-21:
Yep!  Works!  Sorry it was so boneheaded.  I guess I was just following the example given in the readme.  Thanks for the help!

pclov3r, 2017-10-21:
Perhaps the README should include an command example such as that. @argilo suppose this could be easily added.

impala454, 2017-10-21:
I just want to add this is a great piece of software and if I see a way to contribute I will.  Thanks again.

argilo, 2017-10-21:
@impala454 Looks like your problems were resolved, so I'll close this issue.

closed 2017-10-21 by argilo

classicfm, 2017-10-23:
@impala454 two things to do that helped me a lot, even with analog FM: get an old style pair of rabbit ears, as in a dipole antenna. The kind that actually look like rabbit ears. Spread the two antennas as far apart as they'll go. Those dipoles were for VHF-Lo television in the old days, which is hardly used in today's digital world due to impulse noise, but happens to be right near the FM broadcast band frequencies.

You can also try putting your SDR away from the computer using a USB extension cord. I also put mine inside a tin can. It sounds silly but it really did help. You may not need to take this step since some of the dongles being sold are already encased in some aluminum shielding. 

I do this an receive analog FM stereo that is essentially perfect.

impala454, 2017-10-23:
@classicfm thanks for the suggestions.  I do actually have a couple of antennas lying around similar to what you mention (rabbit ears plus a circle in the middle) and will give them a shot.  I do believe though my main issue was simply the misunderstanding of the application syntax.  My FM sounds great with the included antenna, and so far about five or six HD radio stations come in crystal clear as well.  I'm sure eventually I'll be putting a nice one on the roof.