theori-io/nrsc5 #97: enhancement req: demodulator for gqrx

fightingtexasaggie, 2017-10-26:
Is there any interest in making this a demodulator for the popular Linux qgrx SDR system?

argilo, 2017-10-26:
It probably wouldn't be much work to make a fork of gqrx that included nrsc5 as a demodulator. But I doubt it could become part of the upstream project anytime soon, due to the patents covering parts of the NRSC-5 standard.

KeyserSoze1, 2017-10-28:
gqrx is built upon GNU Radio.  If/when decoding is added to gr-nrsc5 then it will be trivial to add to gqrx.  I had a demo that had gqrx with nrsc5 by piping samples out of gqrx into nrsc5 running as a separate process.  It was a little flaky but it was nice to show the side lobe channels for HD radio in the waterfall/spectrum display as the audio was playing.