theori-io/nrsc5 #99: Add ability to select sound output device

tylert, 2017-10-27:
When using nrsc5 on a system with more than one sound output device (e.g.:  headphones plugged into an external USB DAC) there is no easy way to select the desired output device.

I spoke to @argilo briefly about this and it seems like ```output_init_live()``` might be the best place in nrsc5 to add these hooks, perhaps with a command-line option of ```-x``` or ```-a``` or something.

According to https://xiph.org/ao/doc/drivers.html, one may pass to the alsa driver ```hw:0``` for the first soundcard and ```hw:1``` for the second and so on.

awesie, 2017-10-28:
In the interim, you might be able to define the appropriate options in your ~/.libao configuration file.

See https://xiph.org/ao/doc/config.html.

pclov3r, 2017-10-31:
Hum, I wonder how this would affect things and how you'd select devices on Windows.

Suppose it's probably some place in the libao documentation.

chrismrutledge, 2020-04-05:
As an alternative, you can change which device is used as the default device by the system.

Create or edit the alsa configuration file at ~/.asoundrc, or the system wide one at /etc/asound.conf, and use this:

pcm.!default {
    type hw
    card 1
}

ctl.!default {
    type hw           
    card 1
}

Assuming that Card 1 is your USB device since Card 0 is probably the built-in.  If not, use aplay -l to figure out which card is the desired output.