theori-io/nrsc5 #111: ADTS SBR+PS does not play back on devices that should support it

classicfm, 2017-11-16:
I cannot get an SBR+PS stream from nrsc5 to play on an Android device. According to Google's documentation, HE-AACv2 has always been supported in Android. The raw .aac won't play (whereas SBR-only streams from an HD1 channel will play), and any method of remuxing will not change this.

I have tried using mkvmerge to create an .mka file. The situation is the same: HE-AACv1 will play, v2 will not. I have also recently tried using MP4Box from gpac (http://gpac.sourceforge.net/), which has fairly extensive support for MPEG-4 Systems files (i.e. .mp4). It has explicit support for importing ADTS AAC with SBR+PS, and placing these as a track in an mp4 file. 

Apparently, in MP4, you can signal SBR/SBR+PS in a backward compatible way, and in a non-compatible way. Neither works. 

To confirm that the device is not to blame, I can play an HE-AACv2 Icecast/Shoutcast stream without any problems. 

Is it possible that the hdc_to_aac.c file needs to be updated or fixed to handle PS? I have two SBR+PS files that exhibit this issue, from two different stations. I can provide these files on request.

classicfm, 2017-11-16:
It does not make sense to me that the ADTS header would cause problems once the stream is placed in another container, however, I can see some potential issues with the ADTS header from nrsc5, compared here to an HE-AACv2 Icecast stream:

From nrsc5:
`General
Complete name                            : Workspaces/hdc-aac/waer-hd2.aac
Format                                   : ADTS
Format/Info                              : Audio Data Transport Stream
File size                                : 293 KiB
Overall bit rate mode                    : Variable

Audio
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format version                           : Version 4
Format profile                           : HE-AACv2 / HE-AAC / LC
Bit rate mode                            : Variable
Channel(s)                               : 2 channels / 2 channels / 2 channels
Channel positions                        : Front: L R / Front: L R / Front: L R
Sampling rate                            : 44.1 KHz / 44.1 KHz / 22.05 KHz
Compression mode                         : Lossy
Stream size                              : 293 KiB (100%)
`

From a server:

`
General
Complete name                            : cc80s_icy.aac
Format                                   : ADTS
Format/Info                              : Audio Data Transport Stream
File size                                : 391 KiB
Overall bit rate mode                    : Variable

Audio
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format version                           : Version 2
Format profile                           : HE-AACv2 / HE-AAC / LC
Bit rate mode                            : Variable
Channel(s)                               : 2 channels / 1 channel / 1 channel
Channel positions                        : Front: L R / Front: C / Front: C
Sampling rate                            : 44.1 KHz / 44.1 KHz / 22.05 KHz
Compression mode                         : Lossy
Stream size                              : 391 KiB (100%)
`

The version number seems incorrect in the Icecast stream, oddly, but the channel configuration is definitely wrong in the nrsc5 stream. Unless this is causing the remux tools to output bad metadata, I'm unsure that this is to blame for this specific bug. However, it does seem to me that it is wrong and should be fixed nonetheless, because it probably could cause compatibility problems.

awesie, 2017-11-16:
I believe this is expected given the comments on #88. NRSC-5 uses something very similar to DRM PS, which has a different format than the HE-AACv2 PS. I don't know if it will ever be reasonable to directly transcode it from HDC to AAC.

On stations that use SBR+PS, I suggest using the wav output. You can pipe it through ffmpeg to compress it if bandwidth is a concern.

argilo, 2019-02-10:
The transcoding functionality has been removed from nrsc5, so I'm going to close off this issue.

closed 2019-02-10 by argilo