theori-io/nrsc5 #115: Images are being received from all streams

cmnybo, 2017-11-24:
When listening to a station with more than one stream, the album art images are being received from all streams.
Would it be possible to only receive images from the currently playing stream or indicate which stream the image belongs to?

I am trying to write a GUI for nrsc5, and currently there is no way to tell which image belongs to what stream, so I can't show the correct one.

classicfm, 2017-11-24:
I would like to see the program associated data currently printed as debug log messages instead printed as JSON for this kind of application. Perhaps when an AAS file is received the stream it is associated with could be noted in the json stream.

argilo, 2017-11-25:
@cmnybo Album art is mostly undocumented, so it will take some reverse engineering to figure out how to properly process the images. This article mentions that the XHDR ID3 tag is used for image synchronization: http://www.radioworld.com/digital-radio/0014/artist-experience-and-album-art-via-hd-radio/330814

I added some code in #116 to dump out XHDR tags, which may help with reverse engineering. It could also be that the port number determines which program the image belongs to.

I'm excited to see what you come up with for a GUI!

argilo, 2017-11-26:
@cmnybo It looks like the association between programs (HD1, HD2, HD3) and ports is laid out in the "port info" message, which is parsed by the `parse_port_info` function:

https://github.com/theori-io/nrsc5/blob/02cd65b8e4e96a7a573b06364fbb236a915640a8/src/output.c#L404-L462

For my local station, it produces the following:
```
19:34:03 DEBUG output.c:483: 40 01 00 37
19:34:03 DEBUG output.c:497: Found MAJIC 100.3
19:34:03 DEBUG output.c:506: Port 800, type 3, size 265
19:34:03 DEBUG output.c:506: Port 801, type 3, size 265
19:34:03 DEBUG output.c:483: 40 02 00 F2
19:34:03 DEBUG output.c:497: Found 580 CFRA on 100.3 HD2
19:34:03 DEBUG output.c:506: Port 802, type 3, size 265
19:34:03 DEBUG output.c:506: Port 803, type 3, size 265
19:34:03 DEBUG output.c:483: 40 03 00 46
19:34:03 DEBUG output.c:497: Found TSN 1200 on HD3
19:34:03 DEBUG output.c:506: Port 804, type 3, size 265
19:34:03 DEBUG output.c:506: Port 805, type 3, size 265
19:34:03 DEBUG output.c:483: 41 0A 00 18
19:34:03 DEBUG output.c:497: Found HD TMC
19:34:03 DEBUG output.c:506: Port 401, type 1, size 65
19:34:03 DEBUG output.c:483: 41 0B 00 99
19:34:03 DEBUG output.c:497: Found TTN TPEG
19:34:03 DEBUG output.c:506: Port 880, type 0, size 259
19:34:03 DEBUG output.c:506: Port 881, type 0, size 259
19:34:03 DEBUG output.c:506: Port 882, type 0, size 259
19:34:03 DEBUG output.c:483: 41 16 00 B3
19:34:03 DEBUG output.c:497: Found TTN STM
19:34:03 DEBUG output.c:506: Port 810, type 3, size 66
19:34:03 DEBUG output.c:506: Port 820, type 3, size 66
```
The first 12 lines are the associations for the audio programs. I would assume that `40` means "audio program" and `01` means HD1. This particular station only broadcasts station logos, and they are sent on ports 801, 803 and 805.

argilo, 2017-11-26:
I also had a look at the `XHDR` ID3 tags, and observed two types:

* 8-byte tags, e.g. `36 75 4b be 00 02 06 0b`. Only the last two bytes vary. The second-last byte typically increments when a new song starts playing. These messages seems to indicate which album art file should be displayed, but I'm not sure how to link the last two bytes to the file.
* 6-byte messages (I've seen `36 25 c7 d9 01 00`, `36 75 4b be 01 00` and `ff ff ff ff 01 00`). These seem to indicate that the station logo should be displayed.

cmnybo, 2017-11-26:
I also noticed that the second last byte increments when a new song starts on any of the streams.
The first 6 bytes seem to be the same on all of the stations I can pick up like `36 75 4B BE 00 02 B0 66`. 

I modified the log line in output.c to print the port number along with the name for each file received and it does seem to indicate which stream the image belongs to.

cmnybo, 2017-11-26:
From what I can tell, all of the stations around me have two ports on each audio stream.
The album covers seem to be sent on the first port nrsc5 lists. The station logos are sent on the second port.
The image should be updated when the XHDR tag changes. If it has 8 bytes, the album cover is shown, it it has 6 bytes, the station logo is shown.
Can anybody else confirm this on other stations?
I can only receive one station reliably all day, which makes testing difficult.

argilo, 2017-11-26:
@cmnybo There doesn't have to be two ports per audio stream. Here's an example from CKIS-FM:
```
08:22:58 DEBUG output.c:483: 40 01 00 64
08:22:58 DEBUG output.c:497: Found MPS
08:22:58 DEBUG output.c:506: Port 40A, type 3, size 265
08:22:58 DEBUG output.c:506: Port 407, type 3, size 265
08:22:58 DEBUG output.c:483: 40 02 00 6F
08:22:58 DEBUG output.c:497: Found SPS1
08:22:58 DEBUG output.c:506: Port 40B, type 3, size 265
08:22:58 DEBUG output.c:483: 40 03 00 6F
08:22:58 DEBUG output.c:497: Found SPS2
08:22:58 DEBUG output.c:506: Port 410, type 3, size 265
```
This station only shows album covers on HD1.

argilo, 2017-11-26:
I suspect the first byte (`40` or `41`) from `output.c:483` comes from this table: https://hdradio.com/broadcasters/us-regulatory/nrsc-supplemental-information#service_data_types

cmnybo, 2017-11-27:
I built a better antenna today and was able to pick up some stations that don't have any ports or send any images. Some of them are also sending maps and weather alerts on channel type 41 streams.

I would like it if nrsc5 displayed the port number when it prints the name of the received files, since that does identify which stream it belongs to.

argilo, 2017-11-27:
This was resolved in #119, so I'll close for now.

closed 2017-11-27 by argilo