theori-io/nrsc5 #122: Other SDR support

pclov3r, 2017-11-30:
Now this is probably a huge feature request that is complicated but I like to see some other SDR support such as the Airspy, SDRPlay, Etc.

pclov3r, 2018-02-17:
Hopefully this project is still alive. Not seen any activity with it over the last few months.

One this this maybe could be done is via SoapySDR https://github.com/pothosware/SoapySDR/wiki

awesie, 2018-02-17:
This is a feature I've been working on for the past few days, along with some other rearchitecting. I used SoapySDR since I didn't see any alternatives.

One concern with SoapySDR is that it is yet another dependency for people to compile and install. In addition, for each device you want to support (e.g. rtl-sdr, HackRF, Airspy) you need to compile and install additional packages. It looks like the next Ubuntu LTS will have SoapySDR packages, but this does not help MacOS or Windows users.

Another issue that I encountered over the past couple days is that some SDRs do not support arbitrary bandwidths and sample rates. For instance, the HackRF that I am testing has a minimum bandwidth filter of 1.75 MHz, so even though I can sample at 1.49 Msps, the aliasing destroys the signal. So I had to add another decimation filter and sample at 2.98 Msps. From looking at the Airspy library, it may have similar issues, though I don't have one to test.

pclov3r, 2018-02-17:
I own a SDRplay that I am able to test with. It has sample rates of 500Khz, 1 Mhz and 2.048 Mhz and 3 Mhz on the lower end. Would like to see this working :) 

It does sound like there is some issues to get other SDR hardware working with SoapySDR.

ImDroided, 2018-04-24:
I would love to see SoapySDR support with this. I have a PlutoSDR that we are working on custom firmware for and I would love to include this.

mmiller7, 2019-04-19:
Is there an update to this?  I've got a LimeSDR Mini which is supported by SoapySDR and I'm interested.

pclov3r, 2020-06-24:
Haven't seen much happen with this. I think in the past there was ideas to use SoapySDR but that didn't really pan out.

Seems most SDRs don't support arbitrary sample rates that lead to issues and extra work trying to make this happen.

Should we still keep this open?

argilo, 2020-06-25:
It may still be worth doing this. SoapySDR still seems like a good choice; GNU Radio is also looking at incorporating it: https://github.com/gnuradio/gnuradio/issues/2718

SDRs that don't support arbitrary sample rates could be accommodated with the addition of an arbitrary resampler. (liquid-dsp has one that could be used.)

pclov3r, 2020-06-25:
I own a SDRplay RSP2 and a SDRplay RSPdx here for my other SDR hardware that I could test with.

pclov3r, 2020-06-30:
So, With use of https://github.com/SDRplay/RSPTCPServer I was able to use the SDRplay RSPdx with NRSC-5. I'm not sure tho if this should be included as possible option for people to use in the read me?

robman501a, 2020-08-05:
Hi, I own the RSP2pro and would LOVE to use it to decode my local HD Radio stations. Only way I know how to is to unplug the coax all the time and switch it over to my RTL-SDR dongle. By doing this, I'm afraid that I'll wear out the SMA ports.
Any idea how I can listen to NRSC-5 from my local stations without switching SDR's all the time?

robman501a, 2020-08-05:
I know that SDRuno can now load plugins. Maybe you could make a plugin based on this decoder? SDRuno will work natively with the RSP's and the RTL dongles (ext_io required).

pclov3r, 2020-08-06:
> Hi, I own the RSP2pro and would LOVE to use it to decode my local HD Radio stations. Only way I know how to is to unplug the coax all the time and switch it over to my RTL-SDR dongle. By doing this, [...]

You can with this. I posted in the issue with it :)

```So, With use of https://github.com/SDRplay/RSPTCPServer I was able to use the SDRplay RSPdx with NRSC-5. I'm not sure tho if this should be included as possible option for people to use in the read me?

With that said I don't think it's going to yeld any performance increases over a rtl-sdr as it's only going to be 8 bits to maintain support, The RSPdx can do 14-bits up to 6 Mhz.