theori-io/nrsc5 #126: Extracting pictures

cybernard, 2017-12-22:
INFO  decode.c:77: BER: 0.017376, avg: 0.036833, min: 0.002011, max: 0.111192
04:52:25 INFO  output.c:634: Received TMT_0358t7_2_1_20171222_0443_0012.png, port 0810
04:52:25 DEBUG output.c:595: 00000001 7E1CB150 4F328CA0
04:52:25 INFO  output.c:619: File TMT_0358t7_2_2_20171222_0443_0011.png, size 5279, port 0810

Here the log indicated pictures are present, how do I extract them?
Also autonaming and auto-song splitting would be nice.

awesie, 2017-12-22:
You can use the `--dump-aas-files` option to dump those files.

Also, autonaming / autosong splitting of the files does not always make sense. A lot of the files are not related to the song, but are maps, traffic, station logo, etc.

argilo, 2017-12-22:
@cybernard You can also use this add-on program to display the traffic and weather maps: https://github.com/KYDronePilot/hdfm

closed 2017-12-22 by argilo

cybernard, 2017-12-23:
Actually what I really want is to put data like:

 Album: Fix A Drink [Single]
20:15:48 DEBUG output.c:402: Genre: COUNTRY
20:15:49 DEBUG output.c:384: Title: Fix A Drink
20:15:49 DEBUG output.c:390: Artist: Chris Janson
20:15:49 DEBUG output.c:396: Album: Fix A Drink [Single]
20:15:49 DEBUG output.c:402: Genre: COUNTRY

in the jpeg's exif data.
Then embed that into an mp3 or another supported format.