theori-io/nrsc5 #136: multipal SDR issue

ImDroided, 2018-04-24:
I am having a issue when trying to run nrsc5 when there is 2 dongles connected. I have the flight aware ads-b dongle and a RTL-SDR.com V3 dongle connected to my pc. I have to unplug the ADS-B dongle in order to get nrsc5 to load and lock on to a station. I am getting this output 

twitch@kl0p:~/SDR/nrsc5$ nrsc5 -p 10 -d 1 -o - -f wav 95.5 0
21:37:29 INFO  main.c:185: [0] Generic RTL2832U
21:37:29 INFO  main.c:185: [1] Generic RTL2832U OEM
usb_claim_interface error -6
21:37:29 FATAL main.c:285: rtlsdr_open error: -6
RIFF���WAVEfmt (��D����8�qdata���twitch@kl0p:~/SDR/nrsc5$ 
twitch@kl0p:~/SDR/nrsc5$ nrsc5 -p 10 -d 0 -o - -f wav 95.5 0
21:38:14 INFO  main.c:185: [0] Generic RTL2832U
21:38:14 INFO  main.c:185: [1] Generic RTL2832U OEM
usb_claim_interface error -6
21:38:14 FATAL main.c:285: rtlsdr_open error: -6
RIFF���WAVEfmt (��D����8�qdata���twitch@kl0p:~/SDR/nrsc5$ 
twitch@kl0p:~/SDR/nrsc5$ 

I have tried both device index's with no avail. Device 0 is my ADS-B dongle and 1 is my V3 dongle. It seems like my device flag is being ignored because if I stop Dump1090 running with the ADS-B dongle nrsc5 will start up just fine but wont lock a signal due to the filters in my ADS-B dongle blocking FM band. If I unplug the ADS-B dongle nrsc5 works just fine with the V3. Has anyone else experienced this? I am running Ubuntu and any hep on this would be awesome.

billybobgot, 2018-04-26:
It seems that Dump1090 is using the default device which is device 0. Have you tried launching nrsc5 with "-d 1" instead of "-d 0"?(while Dump1090 is running)

billybobgot, 2018-04-26:
Dump1090 could be using both rtlsdr's

ImDroided, 2018-04-26:
Na that's not the case because I can open any other SDR software like gqrx
or rtl_433 while dump1090 is running just fine.

PR #138 (2018-10-28): Open the device specified by the -d argument.
Fixes #136.

The `-d` argument was not working, because its value was not passed into `rtlsdr_open`.

Sorry for the delay in fixing this @ImDroided.
closed 2018-10-28 by awesie