theori-io/nrsc5 #139: Windows: cmake errors building nrsc5

matthewkleinmann, 2018-11-17:
This is the error I get:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
AO_LIBRARY
    linked by target "nrsc5" in directory C:/msys32/home/sdr/nrsc5/src

-- Configuring incomplete, errors occurred!
See also "C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeOutput.log".
See also "C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeError.log".

And here is CMakeOutput.log:
$ cat C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeOutput.log
The system is: Windows - 6.1.7601 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: C:/msys32/mingw32/bin/gcc.exe
Build flags:
Id flags:

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.exe"

The C compiler identification is GNU, found in "C:/msys32/home/sdr/nrsc5/build/CMakeFiles/3.12.4/CompilerIdC/a.exe"

Determining if the C compiler works passed with the following output:
Change Dir: C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp

Run Build Command:"C:/msys32/usr/bin/make.exe" "cmTC_45450/fast"
/usr/bin/make -f CMakeFiles/cmTC_45450.dir/build.make CMakeFiles/cmTC_45450.dir/build
make[1]: Entering directory '/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_45450.dir/testCCompiler.c.obj
/C/msys32/mingw32/bin/gcc.exe    -o CMakeFiles/cmTC_45450.dir/testCCompiler.c.obj   -c /C/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_45450.exe
/C/msys32/mingw32/bin/cmake.exe -E remove -f CMakeFiles/cmTC_45450.dir/objects.a
/C/msys32/mingw32/bin/ar.exe cr CMakeFiles/cmTC_45450.dir/objects.a "CMakeFiles/cmTC_45450.dir/testCCompiler.c.obj"
/C/msys32/mingw32/bin/gcc.exe      -Wl,--whole-archive CMakeFiles/cmTC_45450.dir/objects.a -Wl,--no-whole-archive  -o cmTC_45450.exe -Wl,--major-image-version,0,--minor-image-version,0 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
make[1]: Leaving directory '/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp

Run Build Command:"C:/msys32/usr/bin/make.exe" "cmTC_61ecf/fast"
/usr/bin/make -f CMakeFiles/cmTC_61ecf.dir/build.make CMakeFiles/cmTC_61ecf.dir/build
make[1]: Entering directory '/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_61ecf.dir/CMakeCCompilerABI.c.obj
/C/msys32/mingw32/bin/gcc.exe    -o CMakeFiles/cmTC_61ecf.dir/CMakeCCompilerABI.c.obj   -c /C/msys32/mingw32/share/cmake-3.12/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_61ecf.exe
/C/msys32/mingw32/bin/cmake.exe -E remove -f CMakeFiles/cmTC_61ecf.dir/objects.a
/C/msys32/mingw32/bin/ar.exe cr CMakeFiles/cmTC_61ecf.dir/objects.a "CMakeFiles/cmTC_61ecf.dir/CMakeCCompilerABI.c.obj"
/C/msys32/mingw32/bin/gcc.exe     -v -Wl,--whole-archive CMakeFiles/cmTC_61ecf.dir/objects.a -Wl,--no-whole-archive  -o cmTC_61ecf.exe -Wl,--major-image-version,0,--minor-image-version,0
Using built-in specs.
COLLECT_GCC=C:\msys32\mingw32\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/lto-wrapper.exe
Target: i686-w64-mingw32
Configured with: ../gcc-7.3.0/configure --prefix=/mingw32 --with-local-prefix=/mingw32/local --build=i686-w64-mingw32 --host=i686-w64-mingw32 --target=i686-w64-mingw32 --with-native-system-header-dir=/mingw32/i686-w64-mingw32/include --libexecdir=/mingw32/lib --enable-bootstrap --with-arch=i686 --with-tune=generic --enable-languages=c,lto,c++,objc,obj-c++,fortran,ada --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-time=yes --enable-libstdcxx-filesystem-ts=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw32 --with-mpfr=/mingw32 --with-mpc=/mingw32 --with-isl=/mingw32 --with-pkgversion='Rev2, Built by MSYS2 project' --with-bugurl=https://sourceforge.net/projects/msys2 --with-gnu-as --with-gnu-ld --disable-sjlj-exceptions --with-dwarf2
Thread model: posix
gcc version 7.3.0 (Rev2, Built by MSYS2 project)
COMPILER_PATH=C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/;C:/msys32/mingw32/bin/../lib/gcc/;C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/bin/
LIBRARY_PATH=C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/;C:/msys32/mingw32/bin/../lib/gcc/;C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/lib/../lib/;C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../lib/;C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/lib/;C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTC_61ecf.exe' '-mtune=generic' '-march=i686'
 C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/collect2.exe -plugin C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/liblto_plugin-0.dll -plugin-opt=C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/lto-wrapper.exe -plugin-opt=-fresolution=C:\Users\sdr\AppData\Local\Temp\ccVnRCEg.res -plugin-opt=-pass-through=-lmingw32 -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_eh -plugin-opt=-pass-through=-lmoldname -plugin-opt=-pass-through=-lmingwex -plugin-opt=-pass-through=-lmsvcrt -plugin-opt=-pass-through=-lpthread -plugin-opt=-pass-through=-ladvapi32 -plugin-opt=-pass-through=-lshell32 -plugin-opt=-pass-through=-luser32 -plugin-opt=-pass-through=-lkernel32 -plugin-opt=-pass-through=-lmingw32 -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_eh -plugin-opt=-pass-through=-lmoldname -plugin-opt=-pass-through=-lmingwex -plugin-opt=-pass-through=-lmsvcrt -m i386pe -Bdynamic -o cmTC_61ecf.exe C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/lib/../lib/crt2.o C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/crtbegin.o -LC:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0 -LC:/msys32/mingw32/bin/../lib/gcc -LC:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/lib/../lib -LC:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../lib -LC:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/lib -LC:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../.. --whole-archive CMakeFiles/cmTC_61ecf.dir/objects.a --no-whole-archive --major-image-version 0 --minor-image-version 0 -lmingw32 -lgcc -lgcc_eh -lmoldname -lmingwex -lmsvcrt -lpthread -ladvapi32 -lshell32 -luser32 -lkernel32 -lmingw32 -lgcc -lgcc_eh -lmoldname -lmingwex -lmsvcrt C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/../../../../i686-w64-mingw32/lib/../lib/default-manifest.o C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.3.0/crtend.o
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTC_61ecf.exe' '-mtune=generic' '-march=i686'
make[1]: Leaving directory '/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp'

Parsed C implicit link information from above output:
  link line regex: [^( *|.*[/\])(ld\.exe|CMAKE_LINK_STARTFILE-NOTFOUND|([^/\]+-)?ld|collect2)[^/\]*( |$)]
  ignore line: [Change Dir: C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp]
  ignore line: []
[... 263 lines cut ...]
C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c: In function 'main':
C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:8:19: error: '_Imaginary_I' undeclared (first use in this function)
   return ((int*)(&_Imaginary_I))[argc];
                   ^~~~~~~~~~~~
C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:8:19: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [CMakeFiles/cmTC_7d9a3.dir/build.make:66: CMakeFiles/cmTC_7d9a3.dir/CheckSymbolExists.c.obj] Error 1
make[1]: Leaving directory '/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_7d9a3/fast] Error 2

File C:/msys32/home/sdr/nrsc5/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <complex.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef _Imaginary_I
  return ((int*)(&_Imaginary_I))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Any help would be appreciated.  It had to barf out on the last file....

pclov3r, 2018-11-17:
From https://github.com/theori-io/nrsc5/blob/master/README.md did you setup and compile the libao library? 

It looks like it can't find it.

matthewkleinmann, 2018-11-18:
Thank you for the AMAZINGLY fast reply.  I am pretty sure I did not miss any of the steps.  Here is what my "root" looks like for the project:

sdr@sdr-PC MINGW32 ~
$ ls
fftw-3.3.7  libao  libusb  nrsc5  rtl-sdr

I will go back and rebuild it and get back to you.

Thank you again.

matthewkleinmann, 2018-11-18:
Ahhh, it looks like the build of libao failed.

Configure makes a little noise:
checking for pkg-config... no
*** The pkg-config script could not be found. Make sure it is
*** in your path, or set the PKG_CONFIG environment variable
*** to the full path to pkg-config.
*** Or see http://www.freedesktop.org/software/pkgconfig to get pkg-config.

but I think the real root cause is this make error..

make[2]: Entering directory '/home/sdr/libao/src'
  CCLD     libao.la
.libs/ao_wmm.o:ao_wmm.c:(.text+0x1350): undefined reference to `KSDATAFORMAT_SUBTYPE_PCM'
.libs/ao_wmm.o:ao_wmm.c:(.text+0x1368): undefined reference to `KSDATAFORMAT_SUBTYPE_PCM'
.libs/ao_wmm.o:ao_wmm.c:(.text+0x1387): undefined reference to `KSDATAFORMAT_SUBTYPE_PCM'
.libs/ao_wmm.o:ao_wmm.c:(.text+0x1395): undefined reference to `KSDATAFORMAT_SUBTYPE_PCM'
.libs/ao_wmm.o:ao_wmm.c:(.text+0x13fa): undefined reference to `KSDATAFORMAT_SUBTYPE_PCM'
.libs/ao_wmm.o:ao_wmm.c:(.text+0x1402): more undefined references to `KSDATAFORMAT_SUBTYPE_PCM' follow
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:489: libao.la] Error 1
make[2]: Leaving directory '/home/sdr/libao/src'
make[1]: *** [Makefile:546: install-recursive] Error 1
make[1]: Leaving directory '/home/sdr/libao/src'
make: *** [Makefile:561: install-recursive] Error 1

I understand that libao is not yours, but do you have any thoughts?    

Also, I am running Win 7 64, but have the 32 bit version of MSY2.

awesie, 2018-11-18:
You get that error if you are not linking with ksuser when building libao. Make sure that you ran configure as:

```LDFLAGS=-lksuser ./configure```

matthewkleinmann, 2018-11-18:
I have that, if I echo LDFLAGS I get lkuser.  I will run through it again.
Thank you for your help.

matthewkleinmann, 2018-11-18:
This is too odd, but it compiled cleanly this time.  nrsc5 just finished compiling and passed it's test.  Thank you again for your assistance.

argilo, 2018-11-18:
I'm glad to hear you got it working. I'm closing off this issue as it doesn't appear there was anything wrong with nsrc5.

closed 2018-11-18 by argilo