theori-io/nrsc5 #149: nrsc5_stop doesn't work

argilo, 2019-01-29:
@awesie While playing around with the new API, I noticed that `nrsc5_stop` doesn't work when receiving with an RTL-SDR. I believe this is because the worker thread is blocked in `rtlsdr_read_async` and therefore never checks `st->stopped`. Something else (either `nrsc5_stop` itself or `input_cb`) would need to call `rtlsdr_cancel_async`.

awesie, 2019-01-30:
I had looked up if rtlsdr_cancel_async is thread-safe, but then forgot to use it. I like your idea of calling it from the callback to avoid any potential issues related to threading.

argilo, 2019-01-30:
By the way, I tested the API by prototyping a CLI app in Python. It's available on this branch: https://github.com/theori-io/nrsc5/compare/master...argilo:python-cli-prototype

commit b905594f (2019-01-30): Fix #149.
closed 2019-01-30 by awesie