theori-io/nrsc5 #169: Update readme to include Microsoft Visual C++ 2010 SP1 Redistributable Package

pclov3r, 2019-02-06:
The Microsoft Visual C++ 2010 SP1 Redistributable Package is required to run windows builds of NRSC5. @argilo feel like adding this to the readme? 

It's likely already installed for most users but it should be included in the case for those who don't and get a missing dll error.

It's located at https://www.microsoft.com/en-us/download/details.aspx?id=8328

awesie, 2019-02-06:
My initial reaction is probably not add it to the README because we don't distribute any binary builds.

pclov3r, 2019-02-06:
Maybe not. It seems like it's a niche issue anyways but thought i might bring it up since I had it on a brand new windows install.

pclov3r, 2020-01-25:
I'll go ahead and close this. No longer needs to be open.

closed 2020-01-25 by pclov3r