theori-io/nrsc5 #172: Time synchronization in sync.c cannot tolerate dropped samples

argilo, 2019-02-07:
I noticed that my Windows machine occasionally (once every few hours) stops receiving. The MER drops suddenly and never recovers. After some investigation, I discovered that the problem occurs when the RTL-SDR drops samples. The fine time synchronization code in `sync_process` can only tolerate a timing error of +/- 25 samples. Beyond that limit there is phase ambiguity in the reference carriers, and the synchronizer will happily lock onto the wrong phase.

The problem can easily be reproduced by removing bytes from the middle of an IQ file:

```
xz -d < ../support/sample.xz > ../support/sample
cat <(head -c 20000000 ../support/sample) <(tail -c +20000151 ../support/sample) | src/nrsc5 -r - 0
```

I'll have to think about a good way to detect this situation. The coarse time synchronization in `acquire_process` does not have this problem, so maybe it could be used to check if time synchronization has been lost.

awesie, 2019-02-07:
Should we detect that we are no longer getting valid data and reset the acquire? Or do you think it is possible to more gracefully recover from the situation?

argilo, 2019-02-07:
Yes, I think resetting acquire is the right thing to do, since coarse time synchronization is needed to restore sync.

pclov3r, 2019-02-07:
I'm curious if a RTL-SDR with a TCXO LO changes the situation here?

argilo, 2019-02-07:
No, it does not.

awesie, 2019-02-07:
My first thought is to have frame_process report back to nrsc5.c if fix_header fails (and offset == 0). The other idea would be to make a decision based on the bit error rate calculated in decode.c.

I like using the result of fix_header in frame_process because it is binary whereas we would need to decide a cut-off value for the bit error rate.

argilo, 2019-02-07:
> My first thought is to have frame_process report back to nrsc5.c if fix_header fails (and offset == 0).

I'm experimenting with that very idea as we speak, and I also prefer that over checking the BER.

Another option would be to always perform coarse time synchronization, and reset things if the timing error becomes too large. But this has the disadvantage of being computationally expensive.

PR #173 (2019-02-11): Start over from coarse sync if frame can't be decoded.
Fixes #172.

Since three components are now concerned with synchronization state, I've moved it out into the input struct. There are now three states:

* `SYNC_STATE_NONE` - not synchronized
* `SYNC_STATE_COARSE` - coarse synchronization has completed
* `SYNC_STATE_FINE` - fine synchronization has completed

If the reference subcarriers are not as expected or a frame fails to decode, then the state is set back to `SYNC_STATE_NONE`.

I'm still testing this out and would appreciate feedback. Testing instructions are available in #172. The am [... cut]
closed 2019-02-11 by argilo