theori-io/nrsc5 #177: Audio events are bursty

argilo, 2019-02-10:
Currently, audio events are reported as soon as audio packets are decoded. In the P1 channel, decoding happens one frame at a time, and frames are 1.486 seconds long. This places a heavy burden for buffering on the user of the API.

An alternative would be to have libnrsc5 take on some of the responsibility for buffering. It could store the decoded audio packets and report them gradually as new IQ samples arrive. In fact, this would be necessary if we ever wanted to implement analog blending (#80). The audio PDU header has latency and sequence number fields indicating exactly when audio frames should be played back in relation to the incoming signal.

awesie, 2019-02-12:
When I did some experimenting with this awhile ago (012c1244), I used a cross correlation to align the analog and digital audio, which worked decently iirc. Something to consider if aligning everything accurately (e.g. after demod and filtering) proves difficult.

pclov3r, 2023-07-01:
I know this is an old proposal but I noticed when playing some I/Q files that have fading or multipath present decoding can sometimes be not the best and there will be skipping and or pops. Perhaps this issue is more in line with https://github.com/theori-io/nrsc5/issues/68 If the signal is strong this is not present.

@argilo curious about your thoughts here.

PR #399 (2025-05-05): Implement elastic buffering for audio output
Replaces #343.
Fixes #177.
Fixes #261.
Fixes #330.

The NRSC-5 standard uses a variable-rate audio codec. To account for variations in the instantaneous bit rate, the standard specifies that both the transmitter and receiver use an elastic buffer, so that audio packets enter the transmitter and leave the receiver at a fixed rate, even though the number of packets sent in individual audio PDUs may vary. Unfortunately, the standard does not explain how elastic buffering is implemented, so we are left to guess many of the details.

Up until now, nrsc5  [... cut]