theori-io/nrsc5 #181: Windows: left clicking on power shell or command prompt with NRSC5 running will stop it.

pclov3r, 2019-02-12:
So this is a interesting windows specific issue. Left clicking on a command prompt or power shell window to copy text form NRSC5 will cause it to cease function. 

After you copy your text or back out it will resume.

Text output shows that it dropped sync for some reason.

awesie, 2019-02-12:
It probably causes the writes to stdout/stderr to block until you exit selection mode. Logging is on the same thread as processing, so the sample buffer probably gets full and samples get dropped.

See https://stackoverflow.com/questions/33883530/why-is-my-command-prompt-freezing-on-windows-10.

pclov3r, 2019-02-12:
Hum...

That sounds likely what it is.

argilo, 2020-06-23:
I had a look at this as well, and it seems to be the expected behaviour for console applications in Windows. If it's a problem, you could disable QuickEdit mode or run nrsc5 with the `-q` option to disable log output. While there are a couple workarounds nrsc5 could implement (move logging onto a separate thread, disable QuickEdit mode programatically) I don't think they're worth the effort, so I'll just close off this issue for now.

closed 2020-06-23 by argilo

pclov3r, 2020-06-23:
@argilo I agree with you here. Not worth the effort.