theori-io/nrsc5 #183: Windows extra dependencies

zefie, 2019-02-12:
The guide states you only need to copy libnrsc5.dll and nrsc5.exe, this is incorrect. Following your guide using the msys2-build script, the following need to be copied (all from the same bin folder):

libao-4.dll
libgcc_s_dw2-1.dll
libnrsc5.dll
librtlsdr.dll
libusb-1.0.dll
nrsc5.exe

Only then can nrsc5 run outside of the msys environment.

awesie, 2019-02-12:
That's because the msys build script does not attempt to statically compile everything if the dependencies are already installed, unlike the cross compilation script, and you probably already have those dependencies installed.

I'm torn between updating the README to make this clear and updating the script to ignore the system libraries on the MSYS2 build.

awesie, 2019-02-12:
The latest commit should change the MSYS2 build script to avoid using the system libraries and explicitly statically link.

commit 965be89f (2019-02-12): Don't use system libraries by default on MSYS2.

Fixes #183. This is likely the least confusing default behavior for users.
closed 2019-02-12 by awesie