theori-io/nrsc5 #184: no sound from windows 10

fanlessfan, 2019-02-14:


fanlessfan, 2019-02-14:
I downloaded the binary from http://www.mediafire.com/file/mnd31yr5mqwlmbe/nrsc5.zip, but it has no sound.

output
T:\Downloads\nrsc5>nrsc5.exe 105.9 1
23:41:29 INFO  main.c:192: [0] Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 1488375.071248 Hz
[R82XX] PLL not locked!
23:41:29 INFO  main.c:65: Gain: 0.0 dB, CNR: -0.064775 dB
23:41:30 INFO  main.c:65: Gain: 0.9 dB, CNR: -0.047945 dB
23:41:30 INFO  main.c:65: Gain: 1.4 dB, CNR: 0.028570 dB
23:41:30 INFO  main.c:65: Gain: 2.7 dB, CNR: 0.115323 dB
23:41:30 INFO  main.c:65: Gain: 3.7 dB, CNR: 0.044294 dB
23:41:30 INFO  main.c:65: Gain: 7.7 dB, CNR: 0.021585 dB
23:41:31 INFO  main.c:65: Gain: 8.7 dB, CNR: 0.033728 dB
23:41:31 INFO  main.c:65: Gain: 12.5 dB, CNR: 0.017048 dB
23:41:31 INFO  main.c:65: Gain: 14.4 dB, CNR: 0.136245 dB
23:41:31 INFO  main.c:65: Gain: 15.7 dB, CNR: 0.085661 dB
23:41:31 INFO  main.c:65: Gain: 16.6 dB, CNR: 0.205176 dB
23:41:31 INFO  main.c:65: Gain: 19.7 dB, CNR: 0.222723 dB
23:41:32 INFO  main.c:65: Gain: 20.7 dB, CNR: 0.326507 dB
23:41:32 INFO  main.c:65: Gain: 22.9 dB, CNR: 0.493302 dB
23:41:32 INFO  main.c:65: Gain: 25.4 dB, CNR: 0.761051 dB
23:41:32 INFO  main.c:65: Gain: 28.0 dB, CNR: 1.371540 dB
23:41:32 INFO  main.c:65: Gain: 29.7 dB, CNR: 1.468515 dB
23:41:33 INFO  main.c:65: Gain: 32.8 dB, CNR: 2.681381 dB
23:41:33 INFO  main.c:65: Gain: 33.8 dB, CNR: 3.085399 dB
23:41:33 INFO  main.c:65: Gain: 36.4 dB, CNR: 3.382527 dB
23:41:33 INFO  main.c:65: Gain: 37.2 dB, CNR: 3.777576 dB
23:41:33 INFO  main.c:65: Gain: 38.6 dB, CNR: 3.957990 dB
23:41:33 INFO  main.c:65: Gain: 40.2 dB, CNR: 4.282006 dB
23:41:34 INFO  main.c:65: Gain: 42.1 dB, CNR: 4.507452 dB
23:41:34 INFO  main.c:65: Gain: 43.4 dB, CNR: 4.734265 dB
23:41:34 INFO  main.c:65: Gain: 43.9 dB, CNR: 5.989115 dB
23:41:34 INFO  main.c:65: Gain: 44.5 dB, CNR: 6.075715 dB
23:41:34 INFO  main.c:65: Gain: 48.0 dB, CNR: 6.769104 dB
23:41:35 INFO  main.c:65: Gain: 49.6 dB, CNR: 6.808255 dB
23:41:35 DEBUG main.c:69: Best gain: 496
23:41:43 DEBUG sync.c:243: First block @ 10
23:41:52 INFO  sync.c:248: Synchronized!
23:41:53 DEBUG pids.c:114: Station Name: CFMS-FM

fanlessfan, 2019-02-14:
23:41:53 DEBUG pids.c:114: Station Name: CFMS-FM
23:41:53 DEBUG pids.c:103: Country: US, FCC facility ID: 21
23:41:53 INFO  sync.c:324: MER: -8.487889 dB (lower), -7.766245 dB (upper)
23:41:53 INFO  decode.c:77: BER: 0.156496, avg: 0.156496, min: 0.156496, max: 0.156496
23:41:54 DEBUG pids.c:114: Station Name: CFMS-FM
23:41:54 DEBUG pids.c:103: Country: US, FCC facility ID: 21
23:41:54 DEBUG pids.c:114: Station Name: CFMS-FM

awesie, 2019-02-14:
Two possible things from the log file:

1. Your reception is not great. This may cause audio problems at some point.
2. You are attempting to play audio stream 1. You should try audio stream 0 first and make sure that works; e.g. ```nrsc5.exe 105.9 0```

fanlessfan, 2019-02-14:
thanks @awesie. I put the antenna outside and it works perfect. thx

closed 2019-02-14 by fanlessfan