theori-io/nrsc5 #186: gr-osmosdr support for more than just RTL-SDR

rabarar, 2019-02-19:
any chance you can add gr-osmosdr or SoapySDR to extend the available radios?

pclov3r, 2019-02-19:
SoapySDR has been discussed but it seems there is some concerns about it. 

@awesie want to comment here? Perhaps reading the comments https://github.com/theori-io/nrsc5/pull/131 covers it.

rabarar, 2019-02-19:
Thanks - I read the comments - i think the concern was over what SoapSDR supported - here’s the list:

bladerf
hackrf
rtl
rfspace
mirisdr
And devices from the UHD project:

usrp
umtrx
myriadrf

Seems bigger than RTL… ;)

argilo, 2019-02-19:
Unfortunately it's not as simple as just swapping in SoapySDR for librtlsdr. The nrsc5 demodulator operates at a rate of 744187.5 samples per second. The RTL-SDR does not support that sample rate, and so nrsc5 operates it at 1488375 samples per second and uses a decimator to cut the sample rate in half.

Many of the other SDRs supported by SoapySDR do not support either of those two sample rates, and therefore additional resamplers would be required. Some SDRs (e.g. AirSpy) have a fixed sample rate, and so several stages of rational resampling would be needed to go from the fixed sample rate to 744187.5.

argilo, 2019-02-19:
In addition, nrsc5's input stage assumes 8-bit unsigned samples. That code would need to be modified to support the higher sample resolutions of other SDRs.

argilo, 2019-02-19:
I'm going to close this issue, since it's a duplicate of #122.

closed 2019-02-19 by argilo