theori-io/nrsc5 #191: [Windows]

dabdrmkr, 2019-03-01:
I followed the instructions. I noted a couple of warnings but when I ran the nrsc5.exe -r sample 0 command I just got this output.
Kevin@KPR-Business MINGW64 ~/nrsc5/support
$ xz -d sample.xz

Kevin@KPR-Business MINGW64 ~/nrsc5/support
$ nrsc5.exe -r sample 0
16:37:35 DEBUG sync.c:241: Block @ 3
16:37:35 DEBUG sync.c:241: Block @ 31
16:37:35 DEBUG sync.c:241: Block @ 0
16:37:35 INFO  sync.c:218: Synchronized!
16:37:35 DEBUG pids.c:155: Station Name: KUT
16:37:35 DEBUG pids.c:138: Country: AB, FCC facility ID: 21
16:37:35 DEBUG pids.c:213: Station location: 1.038330, -4.093262, 0m
16:37:35 INFO  main.c:243: MER: 13.5 dB (lower), 12.5 dB (upper)
16:37:35 INFO  main.c:224: BER: 0.000137, avg: 0.000137, min: 0.000137, max: 0.000137
16:37:35 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:35 INFO  main.c:279: Artist:
16:37:35 DEBUG frame.c:533: ignoring partial pdu
16:37:35 DEBUG pids.c:188: Long station name: The University of Texas at Austin
16:37:35 INFO  main.c:243: MER: 13.4 dB (lower), 12.4 dB (upper)
16:37:35 INFO  main.c:224: BER: 0.000183, avg: 0.000160, min: 0.000137, max: 0.000183
16:37:35 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:35 INFO  main.c:279: Artist:
16:37:35 DEBUG main.c:258: Audio bit rate: 64.6 kbps
16:37:35 DEBUG main.c:258: Audio bit rate: 63.7 kbps
16:37:35 INFO  main.c:243: MER: 13.7 dB (lower), 12.4 dB (upper)
16:37:36 INFO  main.c:224: BER: 0.000079, avg: 0.000133, min: 0.000079, max: 0.000183
16:37:36 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:36 INFO  main.c:279: Artist:
16:37:36 INFO  main.c:243: MER: 13.3 dB (lower), 12.1 dB (upper)
16:37:36 INFO  main.c:224: BER: 0.000142, avg: 0.000135, min: 0.000079, max: 0.000183
16:37:36 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:36 INFO  main.c:279: Artist:
16:37:36 DEBUG main.c:258: Audio bit rate: 64.5 kbps
16:37:36 DEBUG main.c:258: Audio bit rate: 63.5 kbps
16:37:36 INFO  main.c:243: MER: 13.2 dB (lower), 12.1 dB (upper)
16:37:36 INFO  main.c:224: BER: 0.000118, avg: 0.000132, min: 0.000079, max: 0.000183
16:37:36 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:36 INFO  main.c:279: Artist:
16:37:36 DEBUG main.c:258: Audio bit rate: 64.1 kbps
16:37:36 INFO  main.c:243: MER: 13.3 dB (lower), 12.5 dB (upper)
16:37:36 INFO  main.c:224: BER: 0.000140, avg: 0.000133, min: 0.000079, max: 0.000183
16:37:36 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:36 INFO  main.c:279: Artist:
16:37:38 DEBUG main.c:258: Audio bit rate: 64.1 kbps
16:37:38 INFO  main.c:243: MER: 13.2 dB (lower), 12.2 dB (upper)
16:37:38 INFO  main.c:224: BER: 0.000164, avg: 0.000138, min: 0.000079, max: 0.000183
16:37:38 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:38 INFO  main.c:279: Artist:
16:37:39 INFO  main.c:243: MER: 13.5 dB (lower), 12.4 dB (upper)
16:37:39 INFO  main.c:224: BER: 0.000118, avg: 0.000135, min: 0.000079, max: 0.000183
16:37:39 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:39 INFO  main.c:279: Artist:
16:37:39 DEBUG main.c:258: Audio bit rate: 64.6 kbps
16:37:41 DEBUG main.c:258: Audio bit rate: 63.6 kbps
16:37:41 INFO  main.c:243: MER: 13.5 dB (lower), 12.3 dB (upper)
16:37:41 INFO  main.c:224: BER: 0.000060, avg: 0.000127, min: 0.000060, max: 0.000183
16:37:41 INFO  main.c:277: Title: You're Listening to Q with Jian Ghomeshi
16:37:41 INFO  main.c:279: Artist:
16:37:42 DEBUG main.c:258: Audio bit rate: 64.1 kbps

Surely something more should be happening or is that a successful test?

Kevin

pclov3r, 2019-03-01:
Trying to figure out the issue your having here. The output look correct and you should be hearing audio.

argilo, 2019-03-02:
Indeed, that output looks correct to me. I'll close this issue for now.

closed 2019-03-02 by argilo

dabdrmkr, 2019-03-02:
Hi Guys,
My error re audio on PC.

I can hear audio now when I play the sample file.

Thanks,

Kevin

From: pclov3r 
Sent: Friday, March 01, 2019 11:07 PM
To: theori-io/nrsc5 
Cc: dabdrmkr ; Author 
Subject: Re: [theori-io/nrsc5] [Windows] (#191)

Trying to figure out the issue your having here. The output look correct and you should be hearing audio.