theori-io/nrsc5 #204: enhancement - recording and getting metadata?

ab1jx, 2019-05-02:
I can probably pipe the output through Sox/rec to record, but is there a way to get the metadata like artist, genre, etc?

I don't think I'm going to find anything here without commercials in it so those would need to be edited out with something like Audacity.  Ever notice with ATSC that commercials don't have captions?  I think you could use that as a switch, but I don't know if there's an equivalent trick with FM.

Looking at https://github.com/Witiko/audacity-bridge/blob/master/audacity-bridge it might be possible to feed stdout from nrsc5 directly into Audacity.

awesie, 2019-05-09:
I'm not really sure what you would want it to do. There is no good way to embed in an audio file a title / author / ... that changes over time. Some formats, such as mkv, support it but iirc the players don't support it.

There are python bindings and an example python app where you have access to the audio data and complete metadata in real time. That would probably be the easiest path forward if you wanted to do something with the metadata.

ab1jx, 2019-05-22:
Really what I'd like to do is automatically harvest songs into MP3 files off the air.  I see titles and artists scrolling by to stdout or stderr while nrsc5 is running.  Just grab those and the wav file and stuff them through LAME so title and artist are both ID3 tags and maybe file names.  Need to chop out the commercials somehow.  The title at least sometimes changes to something generic during commercials which you could detect as being a commercial.

Could I do it from C?  I don't do Python.

I've been working on a program to run rtl_power and sanitize the output into a plot to make it easier to spot nrsc5 signals.  I've only found 2 so far in my location.  As a spectrogram the output from rtl_power has what seems linear output over most of its bins but it tapers off toward the ends.  I hope to automatically stagger 2 scans and interleave the good parts.  And by default it does 88-108 MHz but command line args can specify a different range of frequencies.  Output is to Gnuplot.  It's my project https://sourceforge.net/projects/fmplot/ with some enhancements.  It will scan the FM band in 30 seconds or less and plot to a png file.

salamanders, 2022-01-14:
Had a similar thought - I see the Artist, Track name, and an Image of the album cover show up on the car radio, and it would be very cool to be able to capture those.

argilo, 2025-05-07:
I'll close off this issue, because I think the functionality described here would be best implemented in an external application that interfaces with libnrsc5 through its C or Python API. If you end up building something, I'd definitely be interested in hearing about it.

closed 2025-05-07 by argilo