theori-io/nrsc5 #205: Error compiling on Fedora 30

danburd, 2019-05-31:
I've been trying on and off to compile nrsc5 on Fedora for the last few months. I had no issues compiling and running on Windows but I have not been able to compile yet in Fedora. I am not experienced in compiling software so please forgive me if I am making any basic errors.

I ran 
> git clone https://github.com/theori-io/nrsc5.git
> cd nrsc5
> cmake

I got the following error:

> CMake Error at CMakeLists.txt:181 (message):
>   Built-in libao only supported on Windows.
> -- Configuring incomplete, errors occurred!

I ran
> sudo dnf install libao

the result was
> Package libao-1.2.0-14.fc30.x86_64 is already installed.

In past attempts to compile this, I have fiddled around with the cmake file in order to ensure it knows where my dependencies are located. I still received compilation errors at the later stages.

At this juncture I'd like to ask whether there is a step-by-step compilation guide for Fedora. I understand that the step-by-step in the readme is intended for Ubuntu and Debian so the steps for Fedora may be different. Failing that, are there any general resources or advice I should use or follow in order to compile nrsc5 successfully? I apologize if this is the wrong place to post this but I wasn't sure where else to ask. Thank you very much.

I am running 64-bit Fedora Workstation 30 on an Intel NUC PC.

argilo, 2019-06-01:
You probably need the `libao-devel` package so you get the headers. Do you get farther once you install that?

argilo, 2019-06-01:
@danburd I've added build instructions for Fedora to the readme:

https://github.com/theori-io/nrsc5#building-on-fedora

I'll close off this issue for now. If you're still having trouble getting nrsc5 working, please let me know.

closed 2019-06-01 by argilo

danburd, 2019-06-01:
Thank you. The build instructions in the updated readme worked. I really appreciate your help.