theori-io/nrsc5 #210: [Windows]

WB0OCV, 2019-10-04:


WB0OCV, 2019-10-04:
Attempting to install nrsc5 on 64-bit Windows 10 Pro.  The msys2 installation is up-to-date, but when I attempt to create the nrsc5 executable,  no build files or executables are created.  Listed below is the dialog I receive when I attempt the install per instructions:

MSYS ~
$ pacman -Syu
:: Synchronizing package databases...
 mingw32 is up to date
 mingw64 is up to date
 msys is up to date
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

MSYS ~
$ pacman -Su
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

MSYS ~
$ pacman -S git
warning: git-2.23.0-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) git-2.23.0-1

Total Installed Size:  32.97 MiB
Net Upgrade Size:       0.00 MiB

:: Proceed with installation? [Y/n] n

 MSYS ~
$ git clone https://github.com/theori-io/nrsc5.git
fatal: destination path 'nrsc5' already exists and is not an empty directory.

 MSYS ~
$ nrsc5/support/msys2-build
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do
pacman: unknown option -- a

 MSYS ~
$ cd ~/nrsc5/support

 MSYS ~/nrsc5/support
$ xz -d sample.xz

MSYS ~/nrsc5/support
$ nrsc5.exe -r sample 0
-bash: nrsc5.exe: command not found

MSYS ~/nrsc5/support
$ dir
cli.py                   libao-win-fix.patch  nrsc5.py  win-cross-compile
faad2-hdc-support.patch  msys2-build          sample

MSYS ~/nrsc5/support
$ cd ..

MSYS ~/nrsc5
$ cd build

MSYS ~/nrsc5/build
$ dir

 MSYS ~/nrsc5/build
$

As can be seen above, the script file msys2-build produces an error "pacman: unknown option -- a" and fails to produce any build or install files.

Any ideas would be greatly appreciated.  Thank you.

argilo, 2020-01-25:
Are you sure you followed the instructions correctly? I just tried this with the latest MSYS2 on Windows 10 and it worked fine.

closed 2020-01-25 by argilo

argilo, 2020-01-25:
You might want to try deleting the nrsc5 folder (with `rm -rf nrsc5`) at the start to make sure you aren't using an out-of-date version or have leftover files.

endjuser, 2022-03-27:
That "pacman: unknown option -- a" error will happen if you are running from the regular MSYS2 prompt that opens after install.  You need to start "MSYS2 MinGW" and run the commands from there instead.